homieiot/homie-esp8266 #459: Report switches status in timely manner

dvv, 2017-12-23:
Hello!

I would like the device to report its switches et al. status along with heartbeat telemetry.
How do I do that?

TIA,
--Vladimir

timpur, 2017-12-24:
you should have a look at the examples first 
https://github.com/marvinroger/homie-esp8266/tree/develop/examples

dvv, 2017-12-26:
@timpur the point was in "along with heartbeat telemetry", i.e. without explicit coding that.

bertmelis, 2017-12-26:
You can't. You have to explicitely send the property values.

timpur, 2017-12-26:
@dvv my mad, but ive been thinking of building nodes and components which arnt part of the core Homie, but come with the lib that you can add on to homie, like a heartbeat node. I can share you my lib for such a component if you like.

My heartbeat node allows the users to send a true statement to the device, and the device responds with false a second later, like turning on and off a switch, thus you can tell the device is working properly when ever you feel like.

Tell me your toughts on this idea.

@marvinroger
@bertmelis  
@euphi 
@ThomDietrich