homieiot/homie-esp8266 #464: Adherence to Homie Convention?  (lack of attributes)

awilliam, 2017-12-30:
I'm entirely new to this, so I apologize in advance if I'm misunderstanding.  Reading the Homie convention (https://github.com/marvinroger/homie) and comparing to the TemperatureSensor example, I see:
  temperatureNode.advertise("unit");
which is later set in the setupHandler as:
  temperatureNode.setProperty("unit").send("c");
Monitoring the MQTT traffic from the device, I see:

homie/000001/temperature/$properties', ... (12 bytes))
unit,degrees
homie/000001/temperature/unit', ... (1 bytes))
c
homie/000001/temperature/degrees', ... (5 bytes))
22.00

As I read the convention, 'degrees' should be a property on the node and '$unit' should be an attribute of that property.  These are not separate properties, one is an attribute of the other and $unit should appear as:

home/000001/temperature/degrees/$unit

(and should probably be '°C' rather than simply 'c', nit)

How do I use homie-esp8266 to properly create attributes on properties?  Additionally, how do I even set the $name attribute on a node as the interface only seems to allow setting the $type in the constructor and $properties via setProperty()?  Thanks!

bertmelis, 2017-12-30:
The current esp8266 implementation is not yet conform latest (and not yet final?) convention.

Work in progress.

awilliam, 2017-12-30:
Ok, well that's confusing and disappointing and really seems like it should be mentioned in the known limitations in the docs if not the API reference itself.

timpur, 2017-12-30:
Yes the convention has recently been changed and has not yet be released. When it is it will be implemented in homie 2.2.0

timpur, 2017-12-30:
Actually version names are still undecided. But it will be happing soon. I'm just waiting for final release of the new conversation then I'll implement.

stritti, 2019-04-17:
I think we could close. V3.0.0 is supported now.

closed 2019-04-17 by stritti

lhanson, 2019-10-23:
@stritti Is this documented anywhere? As it stands I don't see any way to set property attributes. I can have a "temperature" node which advertises "units" as a property, but no way to set units as an attribute per the spec. Maybe I'm missing something?

MajorTwip, 2019-10-23:
https://homieiot.github.io/specification/spec-core-v3_0_0/#property-attributes

lhanson, 2019-10-23:
I'm aware that the spec includes property attributes, which is why I'm asking about them. I don't see any functionality in `homie-esp8266` for setting them. From the example in the spec:

```
homie/super-car/engine/temperature/$name → "Engine temperature"
homie/super-car/engine/temperature/$settable → "false"
homie/super-car/engine/temperature/$unit → "°C"
homie/super-car/engine/temperature/$datatype → "float"
homie/super-car/engine/temperature/$format → "-20:120"
homie/super-car/engine/temperature → "21.5"
```
I can currently call `engineNode.setProperty("temperature").send(String(temp));`, but there isn't any equivalent `engineNode.setAttribute("unit").send("°C"));`

MajorTwip, 2019-10-23:
https://github.com/homieiot/homie-esp8266/blob/develop-v3/examples/TemperatureSensor/TemperatureSensor.ino

Sorry for my short answers. But yes, you are right, I couldn't find anything in the doc, but I knew I did it.

lhanson, 2019-10-23:
Awesome, so I set the properties when I call `advertise()`. Thanks!

Maybe I'll submit a PR adding a little explanation to the docs since it wasn't easy to find.