homieiot/homie-esp8266 #465: Cannot build develop branch in PlatformIO

weswitt, 2017-12-31:
Using the current docs the develop branch will not build. The build fails in "ESP Async WebServer" portion:

With lib_deps set to:
  git+https://github.com/marvinroger/homie-esp8266.git#develop
  git+https://github.com/bblanchon/ArduinoJson.git#v5.6.7
  git+https://github.com/marvinroger/async-mqtt-client.git#master
  git+https://github.com/thomasfredericks/Bounce2.git#master
  git+https://github.com/me-no-dev/ESPAsyncTCP.git#master

I get this:

Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\AsyncEventSource.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\AsyncWebSocket.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\SPIFFSEditor.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\WebAuthentication.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\WebHandlers.o
In file included from .piolibdeps\ESP Async WebServer_ID306\src\SPIFFSEditor.h:3:0,
from .piolibdeps\ESP Async WebServer_ID306\src\SPIFFSEditor.cpp:1:
.piolibdeps\ESP Async WebServer_ID306\src/ESPAsyncWebServer.h:35:25: fatal error: ESP8266WiFi.h: No such file or directory
 
With lib_deps set to:
  git+https://github.com/marvinroger/homie-esp8266.git#develop

I get this:
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\AsyncEventSource.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\AsyncWebSocket.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\SPIFFSEditor.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\WebAuthentication.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\WebHandlers.o
Compiling .pioenvs\esp01\lib41c\ESP Async WebServer_ID306\WebRequest.o
Archiving .pioenvs\esp01\lib11b\libBounce2_ID1106.a
In file included from .piolibdeps\ESP Async WebServer_ID306\src\WebRequest.cpp:21:0:
.piolibdeps\ESP Async WebServer_ID306\src\ESPAsyncWebServer.h:35:25: fatal error: ESP8266WiFi.h: No such file or directory
 
I'm sure there is some version of lib_deps that works with the current develop branch, but it does not seem to be documented and I haven't found it anywhere online. If someone knows what is necessary to build the develop branch please let me know.

timpur, 2018-01-01:
Think this related to #444 (were trying to fix this)

kylegordon, 2018-01-09:
#444 is closed now. Maybe worth trying this one again to see if it's fixed for you, @weswitt ?

tripflex, 2018-01-11:
Yeah i'm using PlatformIO and don't have any issues with this.  Go into PlatformIO libraries, and make sure to install `WifiManager` which provides `ESP8266WiFi.h`

timpur, 2018-01-12:
@tripflex thats odd ?? `ESP8266WiFi.h` is park of the core....

euphi, 2018-01-12:
I did not had the problem with WifiManager, but in some of my projects I suddently had to add other internal libs to platformio.ini, e.g. Hash or SPI.

kylegordon, 2018-01-15:
Yes, I think Hash is missing from ESP Async. Have mentioned it here https://github.com/me-no-dev/ESPAsyncWebServer/issues/312.

Aside from that, mostly all resolved?

tripflex, 2018-01-20:
Okay so I just tested this myself on a clean project in PlatformIO ... you must add `#include <Homie.h>` at the top of your `main.cpp` to be able to without throwing this error.

I'm guessing you were just trying to test an empty project without having any homie specific code yet?

weswitt, 2018-01-21:
Thanks everyone. I've moved on and am no longer user PlatformIO or any other IDE. Just too unreliable and unpredictable. I'm using simple tried & true makefiles and am much happier.

timpur, 2018-01-22:
:(, something for us to work towards. Making getting started as easy as we can and improving on our examples and documentation. These things i wish to work on and improve.

fermuch, 2018-01-22:
@tripflex can you share your `platform.ini`? I just created a new project, and I still have problems with ESPAsyncTCP.

luebbe, 2018-01-22:
@fermuch for instance this one: https://github.com/luebbe/homie-ota/blob/master/platformio.ini

You just need this part:
`lib_deps = https://github.com/marvinroger/homie-esp8266.git`

Do you have any "other" libraries inside .piolibdeps in your working directory? If yes, delete .piolibdeps before you build.

Do you have any libraries in the global storage? They could get into the way. I got rid of using the global library storage. All my projects are self contained.

tripflex, 2018-02-14:
@fermuch here's mine, just keep in mind that i'm using 

```
[env:nodemcuv2]
; platform = espressif8266
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino
monitor_baud = 115200
; lib_deps = git+https://github.com/marvinroger/homie-esp8266.git#develop
lib_deps = git+https://github.com/timpur/homie-esp8266.git#v2.1
```

Just keep in mind, that the code above will use the latest staging version of esp8266 for PIO (using GitHub), and @timpur `v2.1` branch (which I was using for testing)

closed 2019-03-30 by stritti

deradam, 2020-05-19:
Seems like this issue is back again. I today installed VSC, Platform.io on a fresh system and created a new project. platform.ini is
~~~
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino

lib_deps = Homie
~~~
And I am getting the same error as described above.
~~~
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.5.1 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20701.0 (2.7.1) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Homie library in registry
Found: https://platformio.org/lib/show/555/Homie
LibraryManager: Installing id=555
Homie @ 3.0.0 is already installed
Installing dependencies
LibraryManager: Installing id=64 @ ^6.11.4
ArduinoJson @ 6.15.2 is already installed
Looking for AsyncMqttClient library in registry
Found: https://platformio.org/lib/show/346/AsyncMqttClient
LibraryManager: Installing id=346 @ ^0.8.0
AsyncMqttClient @ 0.8.2 is already installed
Installing dependencies
Looking for ESPAsyncTCP library in registry
Found: https://platformio.org/lib/show/305/ESPAsyncTCP
LibraryManager: Installing id=305 @ 1.2.0
ESPAsyncTCP @ 1.2.0 has been successfully installed!
Looking for AsyncTCP library in registry
Found: https://platformio.org/lib/show/1826/AsyncTCP
LibraryManager: Installing id=1826 @ ^1.0.0
AsyncTCP @ 1.1.1 is already installed
Looking for Bounce2 library in registry
Found: https://platformio.org/lib/show/1106/Bounce2
LibraryManager: Installing id=1106 @ ^2.1.0
Bounce2 @ 2.53 has been successfully installed!
Looking for ESP Async WebServer library in registry
Found: https://platformio.org/lib/show/306/ESP Async WebServer
LibraryManager: Installing id=306

Downloading  [##########################----------]   72%
Downloading  [###########################---------]   75%
Downloading  [############################--------]   78%
Downloading  [#############################-------]   81%
Downloading  [##############################------]   84%
Downloading  [###############################-----]   87%
Downloading  [################################----]   90%
Downloading  [#################################---]   93%
Downloading  [##################################--]   96%
Downloading  [####################################]  100%
ESP Async WebServer @ 1.2.3 has been successfully installed!
Installing dependencies
LibraryManager: Installing id=305
ESPAsyncTCP @ 1.2.0 is already installed
LibraryManager: Installing id=1826
AsyncTCP @ 1.1.1 is already installed
Looking for Hash library in registry
Warning! Library `{'name': 'Hash', 'platforms': ['espressif8266']}` has not been found in PlatformIO Registry.
[... 49 lines cut ...]

 #include <ESP8266WiFi.h>
                         ^
compilation terminated.
*** [.pio/build/d1_mini/lib2b7/ESP Async WebServer_ID306/AsyncEventSource.cpp.o] Error 1
In file included from .pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/AsyncWebSocket.h:32:0,
                 from .pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/AsyncWebSocket.cpp:22:
.pio/libdeps/d1_mini/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:35:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

 #include <ESP8266WiFi.h>
                         ^
compilation terminated.
*** [.pio/build/d1_mini/lib2b7/ESP Async WebServer_ID306/AsyncWebSocket.cpp.o] Error 1
============================================== [FAILED] Took 13.99 seconds ==============================================
The terminal process terminated with exit code: 1
~~~
Any known issues here? Thanks in advance!