homieiot/homie-esp8266 #466: Failure to list Wifi networks on configuration ui

awilliam, 2018-01-01:
Using homie-esp:
63fb96d (HEAD -> develop, origin/develop, origin/HEAD) Update updater script addressing quirks (#461)

I connect to the homie AP with the device in configuration mode, I get the proper device information, select next to get to the wifi page and see the attached page, nothing to select to move on or configure network.
[image]

I've diagnosed this to an AP within range of me with the following SSID:
"DIRECT-A2-HP DeskJet 3630 series ⸮"
(quotes added)

If I make the following debugging change, wifi scanning works normally:
```
diff --git a/src/Homie/Boot/BootConfig.cpp b/src/Homie/Boot/BootConfig.cpp
index 4fd4412..60a8f1f 100644
--- a/src/Homie/Boot/BootConfig.cpp
+++ b/src/Homie/Boot/BootConfig.cpp
@@ -206,6 +206,10 @@ void BootConfig::_generateNetworksJson() {
   JsonArray& networks = json.createNestedArray("networks");
   for (int network = 0; network < _ssidCount; network++) {
     JsonObject& jsonNetwork = generatedJsonBuffer.createObject();
+    if (WiFi.SSID(network).startsWith("DIRECT-A2-HP DeskJet 3630 series")) {
+      Interface::get().getLogger() << F("Dropped: \"") << WiFi.SSID(network) << F("\"") << endl;
+      continue;
+    }
     jsonNetwork["ssid"] = WiFi.SSID(network);
     jsonNetwork["rssi"] = WiFi.RSSI(network);
     switch (WiFi.encryptionType(network)) {
```
Which results in the following output on each scan:
Dropped: "DIRECT-A2-HP DeskJet 3630 series ⸮"

I don't own this rogue AP, so I can't simply disable it or try to fix this SSID, but this clearly represents a bug in homie-esp8266's handling of interesting SSIDs, making it difficult or impossible to configure devices.

awilliam, 2018-01-01:
Perhaps the issue is not the special characters, but the length of this rogue AP.  This also resolves the issue:

```
diff --git a/src/Homie/Boot/BootConfig.cpp b/src/Homie/Boot/BootConfig.cpp
index 4fd4412..0ff83ad 100644
--- a/src/Homie/Boot/BootConfig.cpp
+++ b/src/Homie/Boot/BootConfig.cpp
@@ -206,7 +206,7 @@ void BootConfig::_generateNetworksJson() {
   JsonArray& networks = json.createNestedArray("networks");
   for (int network = 0; network < _ssidCount; network++) {
     JsonObject& jsonNetwork = generatedJsonBuffer.createObject();
-    jsonNetwork["ssid"] = WiFi.SSID(network);
+    jsonNetwork["ssid"] = WiFi.SSID(network).substring(0,32);
     jsonNetwork["rssi"] = WiFi.RSSI(network);
     switch (WiFi.encryptionType(network)) {
     case ENC_TYPE_WEP:
```

timpur, 2018-01-01:
Very odd, since the max SSID length should only be 32 char. not sure if this should be handled better with the Arduino core or here. Will look into more.

stritti, 2019-04-17:
Similar issue: https://github.com/homieiot/homie-esp8266/issues/351

closed 2020-12-28 by stritti