homieiot/homie-esp8266 #467: Homie 3.0

timpur, 2018-01-02:
Open to any one.

Like to discuss what you think homie 3.0 will be, new feature, more tools, ext. What would you like to see.

Some of my thoughts.
- Under to hood (core changed).
    - New event bus.
    - New task manager.
    - Possibly a add-on aritecture. So you can enable and disable parts of homie core as you need them.
- Conform to Homie 3.0 (TBA)
- Add-ons
    - Battery management
- Compatibility with esp32???(not sure about this yet)

Let's talk.

furyfire, 2018-01-02:
Streamline the usage of JSON. Currently the $properties topic is comma seperated, but the $implementation/config topic is JSON. I don't care which one it is. I just want it to be streamlined. 

Task management is a big plus. Some sort of repository of nodes which you can easily pull from and quickly modify. (Read X sensor on pin Y every Z seconds)

Core addons sounds great if the implementation can be kept clean
Add on ideas
- Battery (readbattery level, sleep intervals etc)
- Debugging (Push debug output over internet, accept external commands etc)
- FileSystem (Push/Pull files to the SPIFFS over MQTT)
- Webserver in normal mode (For advanced user configuration)
- Other IoT protocols implementations (Philips Hue, Alexa, Google Iot, ThingSpeak)

I know these things could be implemented as nodes. I just feel like they would belong better as "add-on" modules that provide offical functionality in line with the Homie protocol.

IVData, 2018-01-02:
Is SSL support a possibility for this?

timpur, 2018-01-02:
@furyfire love it. Thank you for your comment.

euphi, 2018-01-02:
Regarding Add-Ons/Nodes: 

I think most addons will be somehow a HomieNode. So, most work for addons will be to establish a add-on database and an offiicial process how to add the addons there. 

For some cases a extension of the Homie API to support "Add-Ons" will be necessary, e.g. access to write "device properties" for the Battery add-on or more level of property information, e.g. for the FileSystem.

Regarding Debug: I have a [LoggerNode](https://github.com/euphi/HomieLoggerNode) that could be a good starting point for a debug add-on. Until now, it is not directly used by Homie, but it will be easy to adapt, especially if there would be a specific way to set a debug-functions inside homie.
(Or just subclass the Node also from `Serial`).

timpur, 2018-01-02:
@euphi good points, im trying to get in touch with @marvinroger to discus and architecture of these ideas. Might start designing how this might all work. Will start uploading docs and milestone plan soon and a task list and we can all contribute to this next major version.

tripflex, 2018-01-03:
Yeah I agree a collection of addons would be amazing.  I had to scour the internet and GitHub to find any custom nodes I could, and I actually did find a good few of them.  Plus it wouldn't be that difficult to write some basic nodes either, and I wouldn't mind sharing mine as well.

Here's some of the nodes I found on GitHub:
https://github.com/yves-ledermann/homie-esp8266-supplements/tree/master/src/Nodes
https://github.com/luebbe/homie-node-collection/tree/master/src

bertmelis, 2018-01-03:
When (if?) going that direction maybe create An organization github-account to manage all the contributors/maintainers/...

timpur, 2018-01-03:
@tripflex @bertmelis Sounds good 👍

timpur, 2018-01-03:
## Question to all. 

### Homie Internal Node for sensors and controls.

Should homie use nodes or a node for internal things like the stats (wifi signal ...) and other control points like reset and restart and any other we add? Call it `homie-internals` Node ?? This would allow for auto discovery of all the internal sensors and controls. Also means as more are added over time, they can just be auto discovered, which is what we want for our Home Automation software.

This should only be done to sensors and controls, not everything in the internals make sense to move to a node. 

Thoughts.

amayii0, 2018-01-03:
Not using Homie heavily at the moment, as a noob I'm wondering what are possible use cases for internals.

Speaking about OO code, I'd keep private what should me only mine and public what is for everyone.
Internals looks like private, as such not accessible as-is. Could that be actually through some kind of getters setters?
For stuff having clear use cases, exposing that just like any other node looks pretty sane.

ingoogni, 2018-01-03:
When you start generalising the idea of a Homie Internal Node, you quickly arrive at the question "why can I have only one `device` (and `root`) on a ESP8266". Once you have the ability to add different devices it should be made easy for users to 'break out' Internal Nodes. For example I'd like to publish the state a state machine is in and I currently create a `node` for that even though I have to attach that `node` to a `device` where it does not belong

The elegance in Homie for me at the moment is that it takes away the burden of writing and maintaining all the code to get the basics going without getting in the way with bells and whistles I never use.
I definitely see use cases for multiple devices on an ESP even though it adds complexity. I think I would not want this added complexity just for 'some internal data'.

This also  tells how I look at add ons, fine as long as they require little or no added complexity to Homie ESP. An other approach could be to start a separate project that adds/works as a layer between Homie ESP and add ons, keeping it possible to keep working with a "clean core" for those who prefer. Homie ESP just as a reference implementation for the Convention.
Then again, the examples of nodes/add-ons as mentioned by @tripflex don't look invasive at all.

Task management can be a plus, although often relative easy to implement 'ad hoc' due to the repetitive nature of events. Also there are sensors that 'push' data to the ESP and there task management may not be very useful.

Most important to me at the moment is an elegant and concise way to define `root/device/node/property` topologies and the `attributes` that belong to them. I wouldn't mind for example if I had to declare the complete structure(s) in JSON as a global. I've to write the whole thing down as an outline anyway before I start programming.

Oh and documentation. Documentation that not just tells stuff, but documentation that explains.

amayii0, 2018-01-03:
Regarding documentation, it's true that I often had to scratch my head to understand how pretty simple samples work.

Something that still poses a problem for me is how I can easily write a big (like I have 4-5 types nodes, see my repo) project and easily enable some features for given devices.
This seems close to add-ons.

I haven't used homie and even less C++ for some time, could that be done by deriving custom classes? If I'm not mistaken it's possible to inherit from multiple classes at the same time to actually combine features.

About add-ons, could that be as simple as adding a define or include? Maybe inheritance? That would mimic layers of enhancements, going from hardcore Homie to higher abstraction.

ingoogni, 2018-01-04:
Thinkin the concept of Homie Internal Nodes through and taking it into the 'extreme'. An example:
We have a home with a room with a light, lets forget about switches etc.

```
homie/room/light/properties
```

This house a also has the nice feature of a wifi network. On that network:

```
homie/wifi/esp-node-007/properties
```

Now that esp-node happens to be the one that tells us everything about the light and can even control it.

So we sit there happily watching the messages passing by on our screen when suddenly ... what ??

```
home/room/$stats/signal => bytearray(b'92')
home/room/$stats/uptime => bytearray(b'1324')
```

How can that be? Our room is not part of wifi so how can it have an uptime? Wouldn't it be mor logical to have:

```
home/wifi/esp-node-007/$stats/signal => bytearray(b'92')
home/wifi/esp-node-007/$stats/uptime => bytearray(b'1324')

and

home/wifi/$stats/signal => bytearray(b'82')
home/wifi/$stats/uptime => bytearray(b'98324')
```

So, we have two devices, wifi and room, two nodes wifi/esp.. and room/light they are independent even though esp-nade may be the one that controls the light node. Being very strict they should be seen separatly and thus it is required to be able to create multiple devices within homie-esp and it would require big changes in the convention as it mixes things up.

The esp-node in the example should now indeed publish topics on its internals, what other topic, or more specific topics and nodes, does this node service. Publihing this would be a requirement as otherwise the relation between the light and controller would be lost. Information that for example is usefull on a maintenance level.

In the current convention all the device attributes in this view should be either device attributes of the ESP itself or of a network node. Not attributes of a device that the ESP manages.

So, yes Homie Internals Node in my view are 'required' and the place to pulish a lot of data that are now attached to the device it services. If implemented I would opt for a as general as possible way so we can manage more devices with one ESP.

(in the example I chose the ESP to be a node of a network(device) but it is probably easier to see an ESP as a device itself)

timpur, 2018-01-04:
@ingoogni i cant follow. You do know that a device can have many nodes and there is no need for many devices on the same physical device (i hope i miss understood). Are you sure you understand how the homie convention works? I really dont understand your examples.

https://github.com/marvinroger/homie

eg. Top of one of my sketches would look like this.
```c++
// Homie Nodes
HomieNode Node_Temperature("temperature", "temperature");
HomieNode Node_Humidity("humidity", "humidity");
HomieNode Node_Pressure("pressure", "pressure");
HomieNode Node_Light("visiblelight", "light");
HomieNode Node_IR("irlight", "light");
HomieNode Node_UV("uvindex", "uv");
HomieNode Node_Battery("battery", "battery");
```

ingoogni, 2018-01-04:
Oh yes, I'm aware of many nodes :)

Let me try again, lets do it backwards and just look at the MQTT messages. It was rather windy here the last days so lets measure wind speed. So we have a `property` wind speed, what is this a property of? Is it a property of the wind speed measurement device? Is it a property of, lets say climate?

I think it is a property of the latter, so the climate is our node. A node of what? A location on earth. Thus we get the message topology:

```
earth/high-peak/climate/wind-speed
```

Next to the measurement apparatus. It also has properties, battery level, display brightness

```
climate-tools/wind-speed-meter/display/brightness
climate-tools/wind-speed-meter/battery/charge
```
no wind-speed node or property to be seen, even though it can give us the information on the actual wind speed.

Two separate entities that are only 'connected' when we are actually measuring the wind speed. This connection can be only be shown by the wind speed meter, not the other way around. What does the wind know?

```
climate-tools/wind-speed-meter/$provider --> "earth/climate/air-condition/wind-speed"
```

Many of the MQTT message topologies somehow squist the two we have seen above together mixing properties of two sepparate devices into one by adding nodes "in the wrong place". Homie does the same. A Homie-ESP device is our wind meter, it can tell us about the wind, but that wind is not a propertie of the Homie-ESP device.

A practical device that I have is my temperature spider with 8 ds18's. Each one can measure a completely different unrelated process so I'd like to be able to set a different message structure for each sensor, even though they are all attached to the same temperature device.

@timpur As said, it is a view and I hope this clarifies it a bit more

timpur, 2018-01-04:
I think I'm starting to see your view, basically sensors and nodes should be detached from the concept of a device, as a device could hold irrelevant sensors that have no correlation.

bertmelis, 2018-01-04:
I understand the point of view but I don't agree to adapt the Homie convention.  Homie is a *lightweight* convention for IoT using MQTT and I would think -as the name suggests- for home use. (although it would fit industrial use also)

Now if you detach the nodes from the device you still have to somewhere document the link between the node and the device. (which device do you have to check when a node is failing?) Too much complexity imho.

The use case when multiple unrelated sensors are connected to the same device: the data consumer could simple subscribe using a wildcard to `homie/+/temp-node/degrees`.

ingoogni, 2018-01-04:
Yes, you're more than half way there I think.

An ESP carrying a bunch of sensors is of coarse also a device, it can tell us things about it self, what you called "Homie Internal Node", it could tell us things about the state of the attached sensors, "CO2 Sensor warming up, ready in two minutes", but the data the sensors produce belong to another device/node and thus topic/message. With multiple sensors on one 'carrier' you could indeed service multiple detached device nodes.

A completely different, sensor less example. Let's have an MQTT client that is a PID controller/thermostat and it does duty for several processes. It listens to temperature messages of several different processes, your 'sous vide', the heater/cooler in your greenhouse, the fermenter of your beer, your fish tank. It has to send messages for setting valves etc. using topics that fit in the topology for your 'sous vide' etc. 

If you follow the current  Convention it can't be done this way, you'd have to implement multiple PID's or you'd have to train your 'sous vide' and 'sous vide database node' to listen to PID topology messages and create the link between them in yet an other definition.

Just to complicate matters ;)

ingoogni, 2018-01-04:
@bertmelis what I wrote is intentionally "recht in de leer" (sorry, lacking English) and kind of triggeret by previous "feature requests" (Homie Internal Node and auto discovery) where I think that if you implement these in a certain way one could gain 'generality' next to the 'lightweightness'. It would broaden the possibilities without breaking backwards compatibility.
But mainly, it is intended to kick against in the hope something comes of it, as I wouldn't know how to program something like this, or the already existing implementation.

euphi, 2018-01-05:
@ingoogni Lets keep things simple. 

Your example with a PID controller is far beyond what is the intented use of Homie. A PID controller is a function that belongs to the controller, not a homie device. 

For MQTT, you will always have a central instance (the broker), so you can also always have a central controller, too - without a significant loss of reliability or performance.

If you want to have an architecture that allows for direct communication between devices, MQTT (and thus Homie) is not the way to go. For that case, I would propose to use a mesh network and offer micro services on some of the participants (e.g. the "PID controller service").
Drawback of such an architecture is that it is far more complicated to keep control what is happening in your network if some changes are necessary (bug fixes, change of requirements, change of the hardware behind (e.g. a new brewing system with new position for sensors etc).

timpur, 2018-01-05:
Guys lets pin this convo here, id like to keep this about homie 3.0, if more is to be said please create an issue in the respective repo please. 

So more ideas about enhancements of homie 3.0 please ???

ingoogni, 2018-01-05:
@euphi The PID is actually a device I use in my household and "carry around" from the coffee roaster to the sous vide or BBQ. It can service all those 'processes' but not at the same time. A (Homie) MQTT networked version could.
I have no intention for direct communication between devices using MQTT, for that there are faster and better ways.

@timpur moving is fine with me. "Decoupling of Homie-device and its services". Where do you want it, in Homie Convention or here in Homie ESP?

euphi, 2018-01-05:
You already can implement a PIDServiceNode with the current version of Homie and the Homie-ESP8266 implementation. Just create _settable_ properties for the inputs, and a property for the output.

To support multiple channels, use the range feature.

ingoogni, 2018-01-06:
@euphi, yes I'm aware of that, but not what I'm seeking.

```
homie/esp-network/esp-007/battery-level
brewery/plate-chiller/coolant-in/temperature
kitchen/appliances/coffee-roaster/temperature
```
This is not possible with homie-esp and not 'allowed' by the Convention from a single MQTTclient.
After some changes of the code it would probably just boil down to multiple instances of the Convention running on a single device. One required default instance containing the esp-xxxx node with its network ability and its properties, many of those as 'defaults'. 0 or more instances servicing other processes/nodes of other devices.

amayii0, 2018-01-06:
Maybe this is already possible, but can we queue MQTT messages at ESP in case of network issue?
For instance if logging temperature, keeping x minutes/hours of values (maybe a given number of values to limit memory usage).

euphi, 2018-01-06:
@ingoogni This discussion belongs to https://github.com/marvinroger/homie  - and there it may be a proposal for a convention version 4 (or 3.5 or whatever), but not the soon to be released version 3 (#redesign).
So I opened a new issue there.

Just "multiple instances of the convention" is not (yet) possible. According to the convention the device-ID is part of the MQTT topic, so there is no way to add your request following the current revisions of the convention (including to #redesign branch for version 3). Furthermore the highest level topic shall be common for all devices in your network (e.g. 'homie/').

This issue here should be about implementation-specific features for versions 3, so they must be compliant to the [Homie convention #redesign](https://github.com/marvinroger/homie/tree/redesign).

ingoogni, 2018-01-06:
Thanks @euphi

furyfire, 2018-01-07:
Two more items the wishlist

Multiple Wifi Configurations. I have a battery driven device in the car it currently connects to the wifi when it is parked in the driveway but being able to configure two or more hotspots would be a nice feature. 

Possibly remove the "button press to config mode" feature from the Homie Core. Leave it up to the user to implement that feature. The current setResetTrigger does not feel very natural and it would result in one less dependency for Bounce2

timpur, 2018-01-12:
I just was just thinking, did we want to use a node for device sensors and controls (Part of homie) which are built-in? This would mean these sensors are discoverable and not assumed to be part of any spec....

Currently Internal sensors and controls (Part of homie) are assumed part of a spec and not discoverable....

??

ingoogni, 2018-01-14:
$stats/... yes, probably also $online, they are not meta-data but measurement values.

timpur, 2018-01-14:
Yeah for $stats and other measurement values. Not sure about the $online ($online will change with new convention)

ingoogni, 2018-01-14:
Forgot to look at the redesign branch, from that I would add $state, it is not meta-data per se. 

(In my simple Homie-applications I publish the state of the state-machine on a node)

euphi, 2018-01-15:
The reason why $online changed, is to know for sure when all "discovery data" has been sent. 

So it is not a measurement value, but important part of the homie convention.

timpur, 2018-01-18:
Please have a quick look at #473

flozsc, 2018-01-19:
I absolutely like the idea about additions for battery-powered sensors. Especially an integrated option for deep sleep would be great.

timpur, 2018-01-19:
Yes I think adons will be a nice adition. And it might allow us to create a homie core version which is platform dependant and maybe add-ons can be shared for many platforms like esp8266 and esp32 ....

hexxter, 2018-04-05:
From my point of view the support for ESP32 is importent and the addon and more lib of nodes is not necessary. The ESP32 is on the same price level and so much more equipment what for many IOT things required is.

paumito, 2018-05-25:
Hmm, what you guys think of developing homie 3.0 in a RTOS framework? The system could be ported more easily than a bare-metal. Just a thought hehe

timpur, 2018-05-25:
@paumito, i like the idea, but i think its more important that we stick to arduino framework for the users sack ...

paumito, 2018-05-29:
Uhmmm...
So, I was doing some research about the compatibility of the Arduino Framework and FreeRTOS.
I've found this projects:

- https://create.arduino.cc/projecthub/feilipu/using-freertos-multi-tasking-in-arduino-ebc3cc

- https://exploreembedded.com/wiki/Setting_Up_FreeRTOS_on_Arduino

- http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html (toolchain)

-https://www.espressif.com/en/media_overview/news/espressifs-esp32-support-amazon-freertos

timpur, 2018-05-29:
Sorry, those are for arm and the esp32, there is no way to use Arduino and rtos for esp8266

bodiroga, 2018-10-15:
Hi @timpur!

Is there any homie-esp8266 branch that is compatible with the Homie 3.0.0 convention? I'm not talking about the homie-esp8266 3.0 library version, I just need the current library features but compatible with Homie 3.0 convention.

Thanks for your hard work and best regards,

Aitor

bodiroga, 2018-10-19:
Hi again @timpur!

I have created a branch in my own fork repository and I have adapted the library to comply to the latest Homie 3.0.0 version. I have done this for my own personal use, but I will be more than happy to make a PR and contribute back to the amazing work that you and @marvinroger have done! Here you have a link to my repository branch:

https://github.com/bodiroga/homie-esp8266/tree/develop-v3

Do you want me to make a PR against the 'develop' branch of the repository?

Many thanks again for all your help and best regards,

Aitor

bodiroga, 2018-10-29:
Hi @marvinroger!

Reading more carefully the Homie 3.0 specification, I have realized that my implementation of Node Arrays is wrong :disappointed: The array definition has changed significantly, so I need to adapt the library first before submitting the PR. Let's see if the changes are easy, because my C++ skills are very basic :+1: 

Best regards,

Aitor

bodiroga, 2018-10-29:
It's me again @marvinroger :wink: 

Is it true that in the Homie 3.0 version the Node is what can be an array, whereas in the 2.0.1 version it was a property field? The array definition is the worst explained thing in Homie, I should open an issue in the convention repository to improve the documentation (https://homieiot.github.io/spec-core-latest/#arrays). Argh, and the callback mechanism for the settable attribute must be changed also. Step by step.

Let's start clarifying if the _range attribute and isRange() method should be moved from the Property class to the HomieNode class :+1:

marvinroger, 2018-10-29:
@bodiroga it is true. It just felt more natural to have node arrays, rather than properties arrays. 😉

igrowing, 2018-11-11:
@timpur Having `homie-internals` (or any other name for the node) to discover other Homie-powered devices in the network could be useful.

In addition, few useful features:
- Homie location detection - Detect where Homie is to assist with weather/time/DST acquisition.
- Time acquisition from NTP + DST enabling.
- Interrupt to certain change in ADC (rise, fall, change).