homieiot/homie-esp8266 #469: ui_bundle Lite

awilliam, 2018-01-03:
Feature request: A smaller ui_bundle, maybe drop the png image, no font encoding

Rationale: Trying to embed homie-esp8266 in devices that may only have 1M flash, I'm unable to get the UI to work with less than 128K SPIFFS.  A simple smart-plug with power monitoring sketch based on homie-esp8266 is coming in at ~45% of the remaining space and I understand that I need to stay under 50% to enable OTA, so as pretty as the UI is, I expect to rarely use it and would really prefer if I could make use of a 64K SPIFFS layout.

Thanks, great project!

timpur, 2018-01-03:
Yes ill be playing around with the size of the ui bundle in my changed for v2.1, but this might also be good to have a basic build with aesthetics removed to save space.

My goal is also to fix the ui + config in 64kb or less also :)

Also will look into how to reduce the build size of homie (would be nice).

BTopbas, 2018-10-16:
Here is some examples. My programming skills not too much. But i think if you read and look this examples maybe convert homie website to this system.

Note Previous versions of project used the SPIFFS partition to store the static contents for the web interface (HTML, scripts, style sheets, images). Since 1.7.0 all these resources are encoded into a header file (a file with the .h extension) and copied to program memory (PROGMEM) during the build process. This means they are included in the firmware image and only one flash step is required. The overall size of the image is (of course) bigger than just the code but way smaller than the firmware plus the filesystem size. In particular this means that 512Kb devices can now have the full featured web interface.

Here you can read a couple of posts about this procedure:

- [Optimizing files for SPIFFS with Gulp](http://tinkerman.cat/optimizing-files-for-spiffs-with-gulp/)
- [Embed your website in your ESP8266 firmware image](http://tinkerman.cat/embed-your-website-in-your-esp8266-firmware-image/)

I read this in other esp8266 project. Here is the [page](https://github.com/xoseperez/espurna/wiki/WebInterface).