homieiot/homie-esp8266 #470: PlatformIO: Release of espressif8266@1.6.0 breaks build

abbe79, 2018-01-03:
Hi!
Unfortunately the release of platform-espressif8266@1.6.0 breaks the build.
Currently using v2.0.0-beta.3.

Workaround is pinning version to 1.5.0 in platformio.ini:
`platform = espressif8266@1.5.0`

see: https://github.com/platformio/platform-espressif8266/releases/tag/v1.6.0

Thanks for homie, loving it!

bertmelis, 2018-01-03:
Platform, IDE/Editor? I just made a new build using VSCode on Win10 using all up-to-date components.

abbe79, 2018-01-03:
See title, PlatformIO (Home 0.6.0 Core 3.5.1a1).

abbe79, 2018-01-03:
Seems to be library-related:
`Compiling .pioenvs/d1_mini/libc12/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.o
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8
_t) {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long i
nt)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/Users/abbe/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing 
argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]
void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;
^` [...]

luebbe, 2018-01-04:
Isn't this #444 striking again?

bertmelis, 2018-01-04:
@luebbe I think so
See also https://github.com/me-no-dev/ESPAsyncTCP/issues/62

run `pio lib update`

euphi, 2018-01-05:
@ivankravets I think this is a bug in platformio:

- A "fresh" pio installation installs ESPAsyncTCP-1.0.1, because AsyncMqttClient depends on ~1.0.0
  - ---> build fails
- A `pio lib update` updates this to 1.1.3.
  - ---> build is fine again.

So, the "good" thing is that the pio bug fixes a problem in other projects, however, pio should no update ESPAsyncTCP because the semver ~1.0.0 does not allow update to 1.1.x.

I remember that this already has been discussed, so I dont't want to directly open a bug report for now. However, I haven't found such a report!? Did I miss something? If not, I'll file a bug report agains pio-core.

ivankravets, 2018-01-06:
@euphi we don't handle project `lib_deps` when checking for library updates. Please file an issue here https://github.com/platformio/platformio-core

abbe79, 2018-01-06:
Thanks for your help! 
@euphi is right, ESPAsyncTCP-1.0.1 was installed, selecting "Update platforms, packages and libraries" did not update to 1.1.3, but running `pio lib update` did.

closed 2018-01-06 by abbe79