homieiot/homie-esp8266 #471: Size of binary grows by 15-20% with latest version

luebbe, 2018-01-07:
Hi folks,

maybe not the 100% correct place to ask the question here, but I'm hoping for some tips nevertheless.

The binary size of my https://github.com/luebbe/homie-sonoff/ project grew by 20% (from 366544 bytes to 416192 bytes) by switching to the latest version of homie and updating all the dependencies. This is becoming a bit of a problem on sonoff devices which only have 1 MByte of flash.
I don't really understand what is causing this.
Is it possible that the linker just links everything, even unused code from piolibdeps into the binary? 
How can I prevent this from happening?

timpur, 2018-01-07:
im going to be looking into this also with v2.1

luebbe, 2018-01-08:
FYI I made a little experiment with my homie-ota project. In https://github.com/luebbe/homie-ota/tree/playground. I changed platformio.ini so that four builds are made in the following order

- serial1 with no "extra" lib_deps
- ssd1306 with the ssd1306 dependency
- u8g2 with the u8g2 dependency
- serial2 with no "extra" lib_deps (just a clone of serial1)

if you delete .piolibdeps before "pio run", you can see that the ssd1306 and u8g2 lib_deps are downloaded and installed when needed.
The resulting binary size of all four builds is exactly the same (404KByte), which I find mildly confusing. I would have expected more for the "ssd1306" and "u8g2" builds.
Or do the binaries grow in fixed chunks, because only multiples of n KByte can be flashed?

timpur, 2018-01-08:
Good question, I have no idea but when I have time I hope to find out my self. Will pass on anything I learn.

kylegordon, 2018-01-15:
Can confirm, using https://github.com/euphi/Homie_BareMinimum results in a 395K binary, with Homie v2.0.0-beta.3, v2.0.0-beta.2, and Develop

This also allows a one time flash of the Sonoff device I'm using at the moment ( Sonoff Touch with ESP82858 ), and then all future OTA updates fail with an error 400 NOT_ENOUGH_SPACE

Has Homie gotten fat? :-) ( I don't think so, I think it's likely to be a dependency somewhere )

timpur, 2018-01-15:
Biggest change to homie that i know was changing the web server to use the async version. Will look into whats causing the bloat. 

I do know that were still dependant on the core webClient and thats why i requested for a asyncWebClient (https://github.com/me-no-dev/ESPAsyncWebServer/issues/283). The idea being were using as much async stuff as possible and the async version builds on from asyncTCP which is required for mqtt and the asyncWebServer, thus this could reduce firmware size.

- [ ] Look into https://github.com/me-no-dev/ESPAsyncWebServer/issues/283 to aid in firmware size

euphi, 2018-01-16:
For a more modular design (--> homie 3.0) it would be nice to be able to remove the Web server completely. 

I don't need it, because I upload config.json via USB to SPIFFS and the Web server is not used in normal mode. 

As a first step, it could be removed by using compiler flags.

euphi, 2018-01-17:
Homie-BareMinimum:

Old (7.12.17)
`354489	   7928	  31992	 394409	  604a9	.pioenvs/esp01/firmware.elf`

New (17.1.18):
`390705	   9088	  32496	 432289	  698a1	.pioenvs/esp01-latest/firmware.elf`

LDF old:

```
Library Dependency Graph
|-- <Hash> v1.0
|-- <Homie> v2.0.0
|   |-- <ArduinoJson> v5.11.2
|   |-- <AsyncMqttClient> v0.8.1
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |-- <Bounce2> v2.3
|   |-- <ESP Async WebServer> v1.1.1
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <Hash> v1.0
|   |   |-- <ArduinoJson> v5.11.2
|   |-- <Ticker> v1.0
|   |-- <ESP8266HTTPClient> v1.1
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <DNSServer> v1.1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESPAsyncTCP> v1.1.3
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0

```

```
|-- <Homie> v2.0.0
|   |-- <ArduinoJson> v5.12.0
|   |-- <AsyncMqttClient> v0.8.2
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |-- <Bounce2> v2.3
|   |-- <ESP Async WebServer> v1.1.1
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ArduinoJson> v5.12.0
|   |   |-- <Hash> v1.0
|   |-- <DNSServer> v1.1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <Ticker> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266HTTPClient> v1.1
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESPAsyncTCP> v1.1.3
|-- <Hash> v1.0
```

timpur, 2018-01-17:
@euphi which branch/commit for each ?? because homie dev hasn't had major changes in that time frame??

euphi, 2018-01-17:
Yes, it seems to be caused by dependencies.

The difference between 2.0.0-beta3 and develop is quite small.

Unfortunately platformio does not show the full version (https://github.com/platformio/platformio-core/issues/1274).

jowi24, 2018-03-19:
For me, upgrading Espressif 8266 platform from 1.4.0 to 1.6.0 caused an increase of 11%.

kylegordon, 2018-03-26:
I agree, with espressif8266@1.5.0, my binary is 8.9% smaller.

Is this something https://github.com/platformio/platform-espressif8266 can help with?

kleini, 2019-01-24:
Please see #563 for one first compiler flag to reduce the size by just disabling not used features.