homieiot/homie-esp8266 #472: UI Bundle loading failed in client's browser

kisieel, 2018-01-18:
Hey,

I'm playing arround with homie these days an really appreciate all the effort you put into it. Good job!~

However, I'm struggilng with one issue. I want my device to serve a smile web page when in configuration mode for users providing instructions on how to configure the device.

I'm using nodemcu with ESP-12E on board. 

As a first step I'm trying to make the ui_bundle you provided up and running.
I managed to upload it to spiffs without any issues.

It also seems that the file is being found by firmwre:

`Received captive portal request: Redirect: http://192.168.123.1`
`Received captive portal request: UI bundle found`

I set a proper build flag in my platformio.ini file to make sure there's enough space (would be a pitty not to use at least part of my ESP's 4mb memory):

`build_flags = -Wl,-Tesp8266.flash.4m1m.ld`

However when I connect to the AP all I see is a white screen on my desktop and "Page loading failed" message on my iPhone.

Any ideas?

kisieel, 2018-01-18:
Worth to mention that I'm using fresh version of homie:
`lib_deps = git+https://github.com/marvinroger/homie-esp8266.git#develop`

timpur, 2018-01-18:
Nothing comes to mind. Have you pointed the browser at http://192.168.123.1/ when connected to the ap ?

kisieel, 2018-01-18:
Sure thing. I also managed to upload the configuration file with `curl` using the IP, but only then. The one I uploaded to spiffs with:

kisieel, 2018-01-18:
Sorry for bothering you. I'm colsing the ticket now. It was probably some trash in my flash.
I ran `esptool.py -p com4 erase_flash`, reflashed all again and it started to work.

closed 2018-01-18 by kisieel

luebbe, 2018-01-18:
> trash in my flash
that's my quote of the day :)

timpur, 2018-01-18:
Love it :P