homieiot/homie-esp8266 #473: Anyone use the Config Proxy ??

timpur, 2018-01-18:
Wondering if any one uses the proxy in config mode?

What do you use it for and is it needed ?
(Tbh think is has been broken for a while...)

luebbe, 2018-01-19:
tbh, I don't even know what this is ...

furyfire, 2018-01-19:
I recall the reasoning for this feature being that during configuration the web ui can verify that the select hotspot has internet connectivity.
And if needed that internet connection can be used to fetch additional configuration data from a webserver and save it in the config file...

I have a hard time to see a reason for this usecase...

timpur, 2018-01-19:
Yes I don't think it's required and used by anyone. This feature relies on esp core webclient, thus I might look into how removing it shrinks the firmware size.....

tripflex, 2018-01-19:
I'm not currently using it, but was looking at potentially using it based on how I end up deciding to go with initial device provisioning.

For more details, here's the comment that specifies details surrounding this:
https://github.com/marvinroger/homie-esp8266/issues/26#issuecomment-221148701

**;TLDR** - _User is setting up a device, connects to homie wifi network to provision, enters wifi information and device connects to internet.  This then allows commands to be proxied via the homie device to the internet, for things like, logging into a user account and generating a deviceID to use in configuration._

@timpur is there a specific reason you wanted to remove this?  TBH I think it would be good to keep this available, as i'm still trying to figure out how i'm going to implement my setup process ... as the device ID is generated by my server, not by the user entering it themselves.

I haven't yet used the proxy feature, as I was attempting to handle configuration/setup using the mobile app and then just using the API to send commands ... problem is, I still want to have the ability for the user to configure the device via the UI bundle (as a fallback), and the problem is before the device can be fully provisioned, i need to figure out how i'm going to specify the device ID

tripflex, 2018-01-19:
@timpur just saw your comment about esp core webclient ... so yes I can see the reason for removing this, but like mentioned above ... what about those use case scenarios where using the captive portal requires the device ID to be generated by a server, and not manually entered by the end user setting it up?

timpur, 2018-01-20:
Thinking about is, and I wondering if a proxy is the best solution. Wouldn't something like the device being connected to the wifi and can be called directly from the wifi network and services to be configured instead of the ap proxy thing.. idk

Maybe think about how you would want this to work and we'll see if the proxy is the best solution for this remote configuration thingy.