homieiot/homie-esp8266 #474: Initial device provisioning, and handling wifi credential changes/updates

tripflex, 2018-01-20:
Posting this issue to start a discussion and get input from others currently using Homie as well, especially with plans to remove the proxy feature to remove bloat from firmware #473 which is really IMO just a "work around"

_**Initial Device Provisioning**_
   1. How do you currently provision new devices?  Just manually enter the details in captive portal?  What about when you want a user to login to an account to associate a device, and generate an ID to use with the device?

**WiFi AP/Credential Changes or Router Failure**
1. How do you currently handle something like this?  Do you just completely reset the device and configure it again?  

### **Device Provisioning**
So currently to provision a device you have the API endpoints or the captive portal while in configuration mode.  Specifically for anybody that plans to use Homie for something other than basic home projects, there's no simple way to provision a device with a custom device ID.

Case and point, you have a separate web UI for managing devices.  When a new device is setup you generate a unique ID for the device, and then either have to manually enter it in the captive portal, or come up with a hack to work around issues like Android not making calls to API endpoint because the homie device has no internet connection.

This is the reason that the proxy configuration was added, to facilitate the ability to provision a device without requiring the user to know what device ID (or even MQTT config) to enter, and to allow for using Homie in **non-diy** situations ... which IMO would only benefit Homie as it would gain more wide adoption.

### **WiFi AP/Credential Changes**
Say your WiFi router craps out on you and you have to buy a new one, or cable company replaces it ... most end users just use the default included wifi creds, or AP name, so when this happens, your Homie device no longer works and the only way to fix it is to do a full reset on the device. 

Same with if you (or your end client) decides to change their wifi password, or AP name ... most will forget to update any iot devices before doing this, and then end up in the same spot as above, as if AP router failed.

Yes this may be easy for those of us who are technically inclined, but say you wanted to use Homie for clients, you parents, or anybody else who is not technically inclined ... see what i mean?

tripflex, 2018-01-20:
### My specific use case and comments
As I will be using Homie for my home DIY projects, I also plan to use Homie for clients to help with automation.  Specifically for this, I want to allow my clients to be able to setup new devices, without having to manually enter in device ID or MQTT credentials.  I modified the config UI bundle to remove configuring MQTT as those values are statically set, but when setting up a new device, I need some way to reference the device itself to a device stored in my database, without allowing clients to just put in anything they want.

My first attempt at this was to use a mobile device, allow them to configure the initial settings which generates the ID, and then pushes it to the homie device using the API.  This has proved to be problematic as Android does not route traffic to private IP routes if wifi does not have internet, and requires a lot of code just to work around this (and i'm sure iOS will have issues as well).

Before switching to Homie I had created my own implementation using the WifiManager lib for Arduino, which allowed the user to setup the device wifi using captive portal, and then discovering the device using mDNS and SSDP and completing the provisioning.  

Being as though i'm a full time developer and that's only possible due to open source, I decided to give Homie a try so I could contribute and give back to community.

With that said though, my plan is to use Homie on larger projects, outside of basic DIY projects, and **_I think it would be great for Homie, both the project, and the community around it, if the initial provisioning was not targeted just towards DIY projects, which I think could result in much wider adoption and usage of this project _**.

Maybe we could add an API node that could be optional for opening up an API on the homie device, even when it's not in configuration mode?  Thoughts? Ideas? Suggestions?  Complaints?

Sorry about the long rant, there's tons of ideas floating around in my head and just trying to get them out without being a novel to read has proved to be quite difficult.  I look forward to hearing everyone else's thoughts and thank you for your time reading this 😛

timpur, 2018-01-20:
Firstly, why create a unique I'd ? Just use the default built in one ? That would solve your I'd issue ?

Secondly, there are some things in the works for v2.1 which will allow you to switch between config mode and normal mode at will with the homie button to update config via the ui / APIs 
This adition will solve the whole issue of just wanting to quickly update the settings and not reset the device via config mode.
See #433

timpur, 2018-01-20:
That all said there are some good point in there and I'd like to come up with solutions which the community agrees on, let's discuss :).