homieiot/homie-esp8266 #476: Captive Portal Not Loading iPhone and Mac

BTopbas, 2018-01-25:
Hello everyone. I am new with homie. I flashed basic code and flashed ui_bundle.gz to data/homie folder my esp8266. But in iPhone and Mac captive portal not loading. Here is the screenshoots.

[image]

When i go in safari go to 192.168.123.1 url i see this.

[image]

And my macbook(HighSierra) connect esp8266 wifi, opens up captive portal but show nothing. Blank white.

Working only with google chrome and enter manually 192.168.123.1 then i see config web page.

timpur, 2018-01-27:
Odd, will look into

emil2099, 2018-02-17:
To add to this issue, I am getting: 

- Request not found for url: http://192.168.123.1 with Microsoft Edge 38.14393
- ui_bundle.gz download on all Apple devices (iPhone, iPad, MacBook)
- Everything works as expected on Chrome on Windows and MacOS

rgriebl, 2018-02-27:
This is a well known Safari bug on all platforms: Safari can handle gzip compression perfectly fine - as long as the file extension is NOT ".gz". So simply renaming ui_bundle.gz to ui_bundle.htmlgz (plus adjusting CONFIG_UI_BUNDLE_PATH[] in Constants.hpp) fixes the problem. Just tested with iOS and macOS.

timpur, 2018-02-27:
okay good to know, ill see if i can strip with file extension in homie as it responds

commit b7b67767 (2018-03-19): Fix Safari not displaying the config bundle HTML page (Fix #476) (#502)

* Fix Safari not displaying the config bundle HTML page

Safari cannot deal with gzip files that have a "*.gz" file extension.
Simply faking the filename solves the problem though.
closed 2018-03-19 by timpur