homieiot/homie-esp8266 #478: property size limited

Conspir3D, 2018-01-28:
Hi,
there seams to be a bug with the message size of a property.
When i do the following:
```
rgbNodeA.setProperty("config").send("{\"name\": \"" + name + "\",\
\"command_topic\": \"" + basetopic + deviceId + "/switch\",\
\"state_topic\": \"" + basetopic + deviceId + "/status1234567\",\
\"payload_available\": \"true\",\
\"payload_not_available\": \"false\",\
\"availability_topic\": \"" + basetopic + deviceId + "/$online\",\
\"qos\": \"0\",\
\"retain\": \"true\"}");
}
```
the payload of that topic is too huge and will not be send. But the json send here will be much bigger to implement all features.
If the payload is just one character less, it works.

timpur, 2018-01-29:
Firstly, very odd way of using homie (not how it's meant to be used)...... but moving past that there shouldn't be a limit on payload size ... and doesn't seem to be in homie code.

Maybe something to do with free heap, since `String` uses heap not stack.

Conspir3D, 2018-01-29:
I use it as it it meant, as an iot device framework ;)
What i try to do with this large payload is to do auto discovery of mqtt devices by home-assistant.io 

Home assistant provide diffrent types of "sensors" each with a generic and individual config topic as JSON. This payload can become quite big as there are a lot of functional options.

I will investigate deeper into the issue, specially to find a way around that heap/stack issue of large payloads.

thankx a lot for this feedback which give me new ideas to work around this in case it'S not an homie issue.

Conspir3D, 2018-01-29:
When i publish this payload manually via cli, my mqtt broker (mosquitto) handles everything fine. I use gbNodeA.setProperty("config").send() only once at the setuphandler.

Then i tested to define the string as a c string like:
```
char configA[] = "{\"name\": \"Test Device\",\
\"command_topic\": \"homeassistant/light/test-id/switch\",\
\"state_topic\": \"homeassistant/light/test-id/status1234567\",\
\"payload_available\": \"true\",\
\"payload_not_available\": \"false\",\
\"availability_topic\": \"homeassistant/light/test-id/$online\",\
\"qos\": \"0\",\
\"retain\": \"true\"}"; 

rgbNodeA.setProperty("config").send(configA);
```
which doesn't make any difference.

As everything works as expected, the rgbNodeA.setProperty("config").send(configA); command breaks some where because of the largepayload but i have no idea, where and why. Are there any more debugging messages i may turn on?

Why did you mention that this is a very odd way of using homie ? Auto discovery is essential and sometimes needs a more complex (longer) payload than just a few characters. Mosquitto can handle payloads up to 260MB even if this is surly an overkill for MQTT!

timpur, 2018-01-30:
I will look into this, there shouldn't be any limit to payload size. Will fix.

(been thinking about creating a homie addon for home assistant) (#306)

Conspir3D, 2018-01-31:
I don't think that there is a need of any plugin, as this very nice framework have all needed features. From my point of view, there is only the configuration topic issue with the payload size which currently block this usage. Thanks a lot for feedback and this great framework.

Conspir3D, 2018-02-02:
Today i stumbled over another compile error at another project, but reminds me on this issue.

#error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 1000"

Maybe the Problem here is also related to any PACKET_SIZE setting at the PubSubClient library?

timpur, 2018-02-03:
Tested, have no issue with message size. Can send and receive messages at lest 3x of the examples you gave me ....
```c++
Node_Message_Relay.advertise("msg").settable([](const HomieRange & range, const String & value) {
 Node_Message_Relay.setProperty("msg").send(value);
 return true;
});
```

Using home 2.0.0B3 Arduino ESP Core 2.4.0 QIO LwIP v2(536)
This is a bare bone sketch that i use for testing, so maybe use a basic homie sketch and see if your still having issues ?

Conspir3D, 2018-03-02:
Haven't time yet to retest, keep this issue alive e, will report if get back to that Project

timpur, 2018-05-20:
@Conspir3D, no response, closing.

closed 2018-05-20 by timpur

Conspir3D, 2018-05-20:
Vielen Dank für Ihre Nachricht.

Bis zum 25.5.2018 bin ich leider nur eingeschränkt erreichbar. Wenn es Wichtig, oder gar Dringend ist, werde ich mich zeitnah melden. Zwischendurch genieße ich die Berge und werde an Sie denken.

Wenn es UNGLAUBLICH WICHTIG ist, können Sie auch versuchen, mich unter der unten angegebenen Mobilnummer zu erreichen. Aber bitte nicht böse sein, wenn auch dieser Versuch erfolglos sein sollte. Ihr Anrufversuch wird mir in diesem Fall aber von den freundlichen Kollegen der NSA in aufbereiteter Kurzform, mit Aktenzeichen versehen, Auszugsweise direkt übermittelt.

Viele Grüße 

Jan Giebels

Conspir3D GmbH
Berliner Str. 1 – 64354 Reinheim

Telefon +49 6162 9167296
Mobil +49 151 22649657
Fax +49 6162 9167298
E-Mail jan.giebels@conspir3d.com

—

Sitz: Reinheim
Amtsgericht Darmstadt: HRB95337
Geschäftsführer: Jan Giebels

timpur, 2018-05-20:
Sorry, google translation didn't really make sense...