homieiot/homie-esp8266 #479: NodeMCU dead after powering off and powering on

volvorine, 2018-01-29:
I'm using 2.0.0-beta.3.
After I upload the example sketch "LignOnOff" everything seems to be working alright.But if I disconnect the device from the USB and plug it back in nothing seems to happen.No serial readings no nothing.I need to upload the sketch again in order to make it work.In other words, I mustn't cut the power off.
Do you, guys, have any idea what could be the problem?Thanks

volvorine, 2018-01-30:
The board says NODEMCU VER0.0 LSP 12 CORE. I'm powering it from the usb. But once i unplug it from the usb and plug it again it doesn't respond.I need to reupload the sketch again to make it work

timpur, 2018-01-31:
I have a feeling its a mix up with the built in led configured wrongly and its using your tx line instead. After you upload do you get the led blinking as it should and serial output as normal or does it cut out shortly after upload ? and do you get serial output every min from homie sending signal stats?

volvorine, 2018-02-01:
It does blink very fast while uploading, but it doesn't after the upload is finished.However, I do get the serial output every minute.

volvorine, 2018-02-05:
Tried to change board, but still no luck :/

timpur, 2018-02-06:
When you plug it back in, does it connect to mqtt ? Can you see it on the mqtt server ? (Use a mqtt client to watch for message)

volvorine, 2018-02-06:
Nope, it doesn't.Doesn't respond even if I try to ping it.

furyfire, 2018-02-06:
I had trouble with a certain ESP8266 board which left handled the reset button in a different. Whenever I powercycled the device the default state of the button would send Homie back into configration mode.

Try adding
 Homie.disableResetTrigger(); 
in your setup routine

volvorine, 2018-02-06:
Nope, that didn't work as well :(

benzino77, 2018-02-06:
Try this one:
```
Homie.disableLedFeedback();
Homie.disableResetTrigger();
```

volvorine, 2018-02-06:
That didn't work either.Here's a video of my issue:
https://www.youtube.com/watch?v=tUPH2qlqQ58

timpur, 2018-02-06:
Firstly from your vid, as soon as you disconnect the device you kill the serial connection which means you need to close and reopen the serial monitor when you connect the device back in.... 

Secondly, when you get it all working, your led is not working as intended, it should flash while connecting then turn off when connected ... not a major issue atm but a note for you.

as @benzino77 said  add for now till fixed`Homie.disableLedFeedback(); Homie.disableResetTrigger();`, so that we can be sure its not causing an issue... so please leave this in your code till we work this serial issue out then you can remove it.

volvorine, 2018-02-06:
`Homie.disableLedFeedback();` and `Homie.disableResetTrigger();` are added but this doesn't make any difference for now.Appreciate your help, guys.

timpur, 2018-02-06:
just want to check that your closing the serial monitor and reopening it when you connect the device back in?

timpur, 2018-02-07:
So odd