homieiot/homie-esp8266 #480: ESP keeps resetting after the first packet should be sent

chris-iot, 2018-01-29:
Hi,
first of all: Thanks for this amazing framework!

I have an issue that my ESP8266 keeps resetting at the end of my setup handler. It seems like it resets shortly after the code of my setup handler is done or if it enters the loop code for the first time.

My guess is that there are too many packets to be sent in the buffer and this causes an out of memory error, but this is just a guess...
Can anyone of the experts have a look on it?

Thanks.

Here are some more details:
[console_out.txt](https://github.com/marvinroger/homie-esp8266/files/1674059/console_out.txt)
[stacktrace_decoded.txt](https://github.com/marvinroger/homie-esp8266/files/1674063/stacktrace_decoded.txt)
[code.zip](https://github.com/marvinroger/homie-esp8266/files/1674070/code.zip)

I'm using the following versions of libs:
- Homie v2.0.0-beta3
- ESP 2.4.0-rc1 (as all the later don't compile with the latest Homie version, i think this is covered in #477 )
- ArduinoJSON 5.13.0
- AsyncMqttClient 0.8.1
- Bounce2 2.3.0
- ESP AsyncTCP 1.1.0 (i assume this fulfills the requirement from http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.3/quickstart/getting-started/ ...)
- ESP AsyncWebserver 1.1.0

If i did miss something, just leave a message here.
Thanks in advance!

chris-iot, 2018-01-29:
BTW: a fully function based implementation does work (only tested with 3 attached sensors, so it only publishes 3 subnodes, but it works). Code is available at https://github.com/chris-iot/esp8266_iot/commit/a957a433e17b44034d9c0844838c83308005b586 or https://github.com/chris-iot/esp8266_iot/releases/tag/Mainboard_0.1.1

Also the code from the zip file from above is now online. You can find it at https://github.com/chris-iot/esp8266_iot/commit/afca8885aebd7ea9d5497c60f80114163c15a415
Perhaps it is easier to read through the code online instead of needing to download it...

timpur, 2018-01-30:
I belive this is your error. you need to only use the number of devices that have been initialized Use `nextHomie` not `TEMP_MAX_DEVS`. Make sure you check for this through out your code. :)

```c++
for(i=0; i<TEMP_MAX_DEVS; i++){
  homies[i]->homie_setup();
}
```

Other wise pointers will point to weird locations when you try to access them and cause crashes ..... 

Report back if this is not the issue??

chris-iot, 2018-01-30:
Hi timpur, thanks for the reply and the code review. Seems like i had big tomatoes on my eyes :-)
I will fix it in my code, test it soon (most likely tomorrow evening) and give you feedback here.

chris-iot, 2018-01-31:
Hi, the error mentioned in https://github.com/marvinroger/homie-esp8266/issues/480#issuecomment-361445310 was exactly the reason why it failed.
Thanks for pointing me the right direction. I close this one here.

closed 2018-01-31 by chris-iot