homieiot/homie-esp8266 #481: AsyncTCP dependency popping up for espressif8266 again

duststorm, 2018-01-30:
I'm trying to build a project based on Homie
running platformio run -e d1_mini --target upload
I get dependency errors that try to include both ESPAsyncTCP and AsyncTCP (the latter is undesired)
Resulting in fatal error: freertos/semphr.h: No such file or directory
in my platformio.ini I have:
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
lib_extra_dirs = ..\libraries
lib_deps = 
  https://github.com/bertmelis/WifiPrinter.git
  https://github.com/marvinroger/homie-esp8266.git```

There seem to be two dependency issues:
```
...
|   |-- <AsyncMqttClient> v0.8.2
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <AsyncTCP> v1.0.1
...
|   |-- <ESP Async WebServer> v1.1.1
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <AsyncTCP> v1.0.1

```
I managed to get around one of the two by installing an older version of AsyncMqttClient first:
platformio lib install "AsyncMqttClient@0.8.1"

However I cannot find a version of ESP Async WebServer that works. The only two versions available seem to be:
1.1.0, released 4 months ago
1.1.1, released 2 months ago

Using 1.1.0 does not work:
Error: Could not find `ESP Async WebServer` dependency for `Homie` library

So I'm kind of stuck.
How do I debug this issue?

I guess this is kind of like issue #444, except that it still occurs (or are the packages in platformio repos not fixed yet and do I need to grab it from git?)

duststorm, 2018-01-31:
I tried using Homie 2.0.0-beta3 explicitly by putting the following in my platformio.ini
```
lib_deps = 
  https://github.com/marvinroger/homie-esp8266.git#v2.0.0-beta.3
```
It did seem to checkout the right tag:
```
LibraryManager: Installing homie-esp8266
...
Note: checking out 'f605adf8cf32b0bac432d52f1f7a59324281bbe7'.
```

However, I get the same issue.

How can I fix it so that it drags in dependencies that work?

stritti, 2019-04-04:
I think this could be closed. If it is still an issue, please reopen it.

closed 2019-04-04 by stritti