homieiot/homie-esp8266 #483: Is it possible to add Homie event on each mqtt reconnect attempt?

usatenko, 2018-01-30:
onMqttBeforeConnect() method to put right before each try to connect mqtt server
I need it as I have a problem when homie stucks in reconnect loop (broker disconnects it by timeout). Can not see any obvious reason for this.

timpur, 2018-01-31:
Have a look at http://marvinroger.github.io/homie-esp8266/docs/develop/advanced-usage/events/
Does this help?

Have a look at #320. Work with us to fix it and not work around it please. The Goal is to make homie stable !!!!

usatenko, 2018-01-31:
Thank you for the answer. Unfortunately, standard events do not help me, as when Homie is trying to reconnect it turns into some loop and can't do anything else except this trying, I believe it is not good at all even we find right solution, as, in some cases it is good to continue sensors querying and control relays. The part of code I am talking about is:
`
void BootNormal::loop() {
  Boot::loop();
  ...
  if (_mqttReconnectTimer.check()) {
    _mqttConnect();
    return;
  }
  ...
  Interface::get().loopFunction();
  ...
}
`
So, first infinite loop to connect and only then the main loop. And I would love to have an event that can allow me to add some reaction to a situation when my MQTT broker is unavailable for a long time, no matter if it is AsyncMqttClient or lower level library glitch or just the broker is down.
Does it make sense?
I will continue to debug this issue.

timpur, 2018-01-31:
Yeah I can add some more events for V2.1

tema-san, 2018-01-31:
same issue. 
if there is no wifi - ( router hangs for example ) - node stuck ;(
no reaction on inputs... button to turn on/off light is not working. 
is it possible to do main loop indepentent from ~Wifi/mqqt ?

nerdfirefighter, 2018-02-02:
looking at the existing code, it should be do able. a few assumptions. 
my understanding of the .send function is it will return 0 if it is unable to send a message an return a message id (>0) if its able to send the message. So if you care to know if the send was successful you should check the return code from the function.

if thats fine, then we may be able to do:

void BootNormal::loop() {
  Boot::loop();
...
  if (_mqttReconnectTimer.check()) {
    _mqttConnect();
  }

  if (Interface::get().getMqttClient().connected()) {
...
// Things that require us to be connected.
}
// regardless of if we are connected or not, run the loop
// This could also have a param that allows you to tell the loop
// if you want it to run regardless of connect or if it has to wait until its connected.
Interface::get().loopFunction();
...
}

usatenko, 2018-02-03:
OK, I found a solution for me, I just can use main loop (not homie loop) and check all the things there. However, last days I did not see any MQTT connection freezes, so, this is so strange error.

stritti, 2019-04-17:
Could this issue be closed?

closed 2019-04-26 by stritti