homieiot/homie-esp8266 #485: Can't config using the web interface.

htodd, 2018-02-08:
I was following an online guide to "fix" a wifi-controlled outlet but the outlet they suggested no longer seems to be available at Home Depot or Walmart.
http://www.hagensieker.com/blog/page/?post_id=44&title=esp8266-hack-of-inexpensive-wifi-outlet

I got a Wuudi from Amazon, and I found it has an ESP13 instead of the usual ESP8266 in the guide. After some iterative messing around, I think I figured out what pins go where and I loaded the Arduino program on it. I can get the button to switch the relay/LED on and off and I see the Homie-XXXXX hotspot and my Ubuntu-based Firefox tries to connect but never completes. I also tried the curl example I found online, and that didn't work.

Looking at the interface with wireshark, I see GET requests for /success.txt that succeed, and GET requests for / which return "not found".

Any ideas what I'm doing wrong?

Thanks.

timpur, 2018-02-08:
We need some more details which homie version you using, do you get serial output from homie ? do you get redirected to a web page when you connect to the homie hotspot with your phone (or com) ?

htodd, 2018-02-08:
I tried master as well as the 2.0 tag. I am not redirected to a web page (the latest Firefox on Ubuntu 16.04LTS, all updated).

I'm not sure about how to get serial output on an Arduino. Do I shutdown the Arduino software and connect to the serial port?

timpur, 2018-02-08:
I'm sorry this is beyond homie support, you should read up on how to use the esp8266, but considering you got the firmware on there your no that far away. I recommend opening the serial monitor in the Arduino ide.

htodd, 2018-02-09:
Took a bit but got the serial output running. 

I go to a wifi connection, load http://marvinroger.github.io/homie-esp8266/configurators/v2/, and then go to Homie-xx xx xx xx xx xx. What I see (repetitively) is:

Triggering Wi-Fi scan...
___ Wi-Fi scan completed
Received captive portal request: Redirect: http://192.168.123.1
Received captive portal request: UI bundle not loaded. See Configuration API usage: http://marvinroger.github.io/homie-esp8266/
Received captive portal request: Redirect: http://192.168.123.1
Received captive portal request: UI bundle not loaded. See Configuration API usage: http://marvinroger.github.io/homie-esp8266/
Received captive portal request: Redirect: http://192.168.123.1
Received captive portal request: UI bundle not loaded. See Configuration API usage: http://marvinroger.github.io/homie-esp8266/
Received captive portal request: Redirect: http://192.168.123.1
Received captive portal request: UI bundle not loaded. See Configuration API usage: http://marvinroger.github.io/homie-esp8266/
Received captive portal request: Redirect: http://192.168.123.1
Received captive portal request: UI bundle not loaded. See Configuration API usage: http://marvinroger.github.io/homie-esp8266/

timpur, 2018-02-09:
loading the UI setup web site beforehand, is a known issue see #446. Recommend you load up the UI setup Site onto the device its self via SPIFS see http://marvinroger.github.io/homie-esp8266/docs/develop/advanced-usage/ui-bundle/

Else you can always load up a config.json file directly and not worry about the ui setup site... (much quicker) http://marvinroger.github.io/homie-esp8266/docs/develop/configuration/json-configuration-file/ 

this CORS issue is fixed in my form v2.1 which will soon be implemented here. (https://github.com/marvinroger/homie-esp8266/pull/458)

htodd, 2018-02-09:
That worked great, after a bit of messing around (which seems always necessary on Arduino). 

I tried going through all the known issues, but there are a lot of them. If you don't fix this, you may think about adding it to your FAQ.

Thanks for the help!

timpur, 2018-02-10:
Well trying to fix all these issue in v2.1

But also the docs need updating and fixing up like you said

closed 2018-05-23 by timpur