homieiot/homie-esp8266 #486: Hostnames are not working correctly

JorinL, 2018-02-10:
I use five devices actually with homie 
- 1x Wemos D1
- 4x SonOffS20

The Wemos D1 works fine and the hostname "Roombot-T-1000" ist working.
On the SonOffs every device has a Number  SonOffS2-1, SonOffS2-2, SonOffS2-3 and SonOffS2-4.

Currently if i look into my WLAN I see: one SonOff named "u" and two "s" and the other is not showing a hostname.

[image]

Config of all 4 is the same beside the number in the name:
`{
  "name": "SonOffS20-2",
  "device_id": "sonoffs20-2",
  "wifi": {
    "ssid": "Wifi-Name",
    "password": "XXXXXXXXXXXXXXX"
  },
  "mqtt": {
    "host": "192.168.XXX.XXX",
    "port": XXXX,
    "base_topic": "homeland/haushalt/elektrik/sonoff/",
    "auth": true,
    "username": "XXXXXXXXXXXXXX",
    "password": "XXXXXXXXXXXXXXXXXXX"
  },
  "ota": {
    "enabled": true
  }

}
`

The Roombot-T-1000's config looks like:

`{
  "name": "Roombot-T-1000",
  "device_id": "Roombot-T-1000",
  "wifi": {
    "ssid": "Homeland_Security",
    "password": "6887612432920345"
  },
  "mqtt": {
    "host": "192.168.0.8",
    "port": 1883,
    "base_topic": "homeland/haushalt/roomba/",
    "auth": true,
    "username": "roomba",
    "password": ";kDr-U[5mj&p6BrX#Rn5"
  },
  "ota": {
    "enabled": true
  }

}`

Do i have an error like -  "#define BRAND "sonoff-s20"" is at all 4 the same? Or what could it be - bug? Config is on all 5 devices applied by curl.
Thank you!

JorinL, 2018-02-10:
I got it now that way, that no one of the four SonOffs exposes a hostname - did that by taking them all offline and deleting the leases. But normally I would expect the same way like it works with de Wemos D1 and Homie. Something seems to be not working correctly.

JorinL, 2018-02-13:
Okay -closed-

Always trust linux and not all this crap :-)

closed 2018-02-13 by JorinL

timpur, 2018-02-13:
tbh never understood the issue :P

JorinL, 2018-02-13:
Somehow i had crap in my wifi network and the hostnames did not show correctly. 
But i could prove them by cmd "host sonoff-s20-1" and voilà suddenly the rest of the network knows the right name.....

So nothing with homie... :-)