homieiot/homie-esp8266 #487: loopHandler not called until Homie is connected

jhergeth, 2018-02-11:
My device registers a loopHandler with setLoopFunction() and does all its work within that loopHandler.
Now I want my device in a "boot" mode until the WLAN is connected.
Having this code:
```
void loop() {
    if(Homie.isConfigured() && !Homie.isConnected()){
        layt->setBootMode(bm);
        loopHandler();  // this is neccessary !!!
    }
    else{
        layt->setBootMode(0);
    }
    Homie.loop();   
    delay(1);
}
```
I needed to add the line with "loopHandler()" in the main loop to be called when the WLAN is not connected in order to have my code in loophandler() beeing called.

In my opinion Homie.loop() should call the handler which has been set using setLoopFunction() in all cases. Or, did I miss something in the documentation?

With regards
Joachim

cecilios, 2018-02-11:
This is related to #354

Nzbuu, 2018-02-12:
I just put the code that I want to run always in the loop function and the code that requires an MQTT connection in the loop handler.

euphi, 2019-03-10:
Closed with #490.

closed 2019-03-10 by euphi