homieiot/homie-esp8266 #488: Can't set default value for HomieSetting<bool> ?

euphi, 2018-02-11:
Based on v2.0.0-beta.3, it seems that it is not possible to set a default value for a HomieSetting<bool>.

Although  `setDefault(false)` has been called inside a constructor of a global object, Homie complains that a mandatory setting is missing and enters configuration mode.
Other settings (of other types) work fine:

Output (including some debug output added to Homie) :
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
ThermostatNode::ThermostatNode()     <-------- ctor of this object calls setDefault of the "Temperature offset" HomieSetting<double>
TouchCtrl::TouchCtrl()               <-------- ctor of this object calls setDefault of the "Print touch data " HomieSetting<bool>

Starting..        <------- here Homie.setup() is called
Checking setting Temperature offset of type double [opt]   <---- this setting is optional (_required == false)
Checking setting loglevel of type string [opt]
Checking setting Print touch data of type bool [req]    <---- this setting is required (_required == true)
Missing bool Print touch data
✖ Config file is not valid, reason: Print touch data setting is missing
Configuration invalid. Using CONFIG MODE
💡
🔌 Booting into config mode 🔌
Device ID is 2c3ae80e57b4
AP started as Homie-2c3ae80e57b4 with IP 192.168.123.1
```

TouchCtrl.cpp
```C++
HomieSetting<bool> TouchCtrl::printSerial ("Print touch data", "flag, if to print raw touch data to Serial");

TouchCtrl::TouchCtrl(): serialPrintData(false) {
	Serial.println(__PRETTY_FUNCTION__);
	printSerial.setDefaultValue(true);
}
```

based on TouchCtrl.h

```C++
class TouchCtrl {

public:
	TouchCtrl();
private:
	static HomieSetting<bool> printSerial;
};
```

Has anyone any idea what could be the problem?

timpur, 2018-02-11:
Everything looks right, but can you try a bool setting implemented normally and see if you have the same issue ?

euphi, 2018-02-11:
It seems to be related to the initialization order... If I call `setDefaultValue(false)` later (e.g. in `setup()` it works as expected.

It seems that the constuctor of my `TouchCtrl::printSerial` is called after the constructor `TouchCtrl::TouchCtrl()` (there is only one instance)

However:
* The "ThermostatNode" is constructed before and works fine
* I'm quite sure that static members are constructed before any non-static object.

timpur, 2018-02-12:
Try this (2nd answer)

https://stackoverflow.com/questions/211237/static-variables-initialisation-order

timpur, 2018-03-12:
@euphi any progress ?

euphi, 2018-03-12:
Not really, just more confusion....

The stack overflow question/answer did not help, as I was already aware of the problem in general and its potentional solutions. However, playing around with initialization order is complicated.

timpur, 2018-03-12:
I'll try have a look when I can