homieiot/homie-esp8266 #489: Trying to load the configuration gave me an exception

htodd, 2018-02-12:
Not sure if you can get anything from this but here's what I had from the serial interface:

Wi-Fi scan completed
Received config request

Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40024db1 depc=0x00000000

ctx: sys 
         sp: 3ffffb30 end: 3fffffb0 offset: 01a0

>>>stack>>>
           3ffffcd0:  4021ada4 00000000 00000027 00000000  
3ffffce0:  3fff02d4 40254a6e 3fff64e2 00000228  
3ffffcf0:  00000228 00000008 3ffffdc0 401004d8  
3ffffd00:  3fff6462 40104456 3fff27ac 4020c3a2  
3ffffd10:  00000036 3fff2160 3fff27ac 3fff0278  
3ffffd20:  3ffffd90 3fff02d4 3ffffdc0 4020c557  
3ffffd30:  00000010 00000010 00000000 40220500  
3ffffd40:  3fff2160 00000001 3ffe954f 402084f0  
3ffffd50:  402282cf 3fff642c 3ffffd94 4020c78c  
3ffffd60:  3fff0178 3fff02d4 3fff02d4 40220684  
3ffffd70:  00000010 3fff6bac 3fff6bac 3fff0278  
3ffffd80:  3fff0178 3fff02d4 3fff4874 4020c838  
3ffffd90:  3ffffdc0 3fff6bdc 3ffffdc0 4021db0a  
3ffffda0:  3fff0178 3fff02d4 3fff4874 4020cbc6  
3ffffdb0:  00000005 00000000 00000020 40100ec6  
3ffffdc0:  3ffe8dd0 401042c3 00000000 00000228  
3ffffdd0:  4010320d 00040000 7fffffff 00000000  
3ffffde0:  00000022 4010320a 00040000 40101db2  
3ffffdf0:  3ffee9e8 40102136 00000000 00000000  
3ffffe00:  00007fff 126f50ce 00002200 3fff64e2  
3ffffe10:  3fff6004 3fff4874 3fff4874 4020ccc0  
3ffffe20:  3ffffe30 00000108 00000937 4021c8c0  
3ffffe30:  402311d6 00000030 3fff4874 402206c8  
3ffffe40:  00000010 00000000 3fff27df 402210d4  
3ffffe50:  00000000 000000d1 3fff4874 4021ad14  
3ffffe60:  00000026 00003c20 000000d0 3ffed050  
3ffffe70:  3ffee238 3fff3c6c 00000008 00000030  
3ffffe80:  0000000f 00000eb2 3ffeaef0 3ffea824  
3ffffe90:  00000000 3fffc6fc 000000a0 00000000  
3ffffea0:  3fff6004 3fff649c 00000000 4021adb5  
3ffffeb0:  3fff007d 00000000 3fff1fa4 40214448  
3ffffec0:  3fff50cc 00000001 3fff4eec 00000001  
3ffffed0:  00000036 00000000 00000000 3fff1fc4  
3ffffee0:  3fff50cc 3fff1fa8 3fff1fa4 4021448c  
3ffffef0:  3fff50cc 3fff1fa8 3fff1fa9 4022acc4  
3fffff00:  017ba8c0 1a940162 a7a8263a 00000000  
3fffff10:  00000000 0000005d 3fffff80 3fff64ba  
3fffff20:  3fff2160 3fff649c 3fff2204 4022ebe9  
3fffff30:  00000014 000005ed 000005ed 3fff2160  
3fffff40:  3fffdc90 3fff27ac 3fff518c 3fff4054  
3fffff50:  00000008 3fff2160 3fff649c 40228249  
3fffff60:  3fffdc90 3fff27ac 3fff518c 401045cc  
3fffff70:  4024550e 3fff27ac 3fff518c 4024553c  
3fffff80:  3fff64ac 3fff649c 00000002 00000000  
3fffff90:  40240473 00000000 3fff518c 402475d7  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,0)

 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset

timpur, 2018-02-12:
Was this a once off from flashing ?
Did resting the board and trying again fix this ?

htodd, 2018-02-13:
Ah, reloaded and it no longer panics.

��〽 Sending statistics...                                                        
  ��• Wi-Fi signal quality: 94%                                                   
  ��• Uptime: 60s                                                                 

Still haven't gotten the mqtt messages to the plug though.

timpur, 2018-02-13:
> Still haven't gotten the mqtt messages to the plug though.

Don't know what you mean by that.

Looks like to me everything is fine.

htodd, 2018-02-13:
Sorry, I wasn't clear enough with my follow-up question. Looks like it's linked but I'm unsure about what mqtt command to send to turn the switch on and off. Can you point to the documentation or whether I can decipher the string by looking at the source code?

timpur, 2018-02-13:
Not sure what exmple your using .....
But from most examples a Boolean value looks like this in string 

`bool on = (value == "true");`

htodd, 2018-02-14:
Can you tell me if this string is correct? 

$ mosquitto_pub -d -t '/homie/<mac_addr>/plug/on/set' -m true -u "<username>" -P "<password>"
Client mosqpub/10671-htodd-HP- sending CONNECT
Client mosqpub/10671-htodd-HP- received CONNACK
Client mosqpub/10671-htodd-HP- sending PUBLISH (d0, q0, r0, m1, '/homie/6001941a3a26/plug/on/set', ... (4 bytes))
Client mosqpub/10671-htodd-HP- sending DISCONNECT

timpur, 2018-02-14:
Looks correct, but is your homie device using its Mac as the Id or did you set one ?

amayii0, 2018-02-14:
An easy way to trace debug MQTT in windows is with MQTT. FX. Not sure for other platforms but you should be able to find alternatives.
I use it often as it as live monitoring, highlights and so on

htodd, 2018-02-14:
Got it working. To be exact, the "name" in config.json is what <mac_addr> should be, and using colons was giving me grief.

timpur, 2018-02-14:
Yes that's how homie works, but seems we need to improve on our docks, there old and missing some of these details.

Will close for now, but might ask you later to review some doc improvements later on and help us fill the gaps.

closed 2018-02-14 by timpur