homieiot/homie-esp8266 #491: [Featurerequest] Restart/Reset without deleting configuration

JorinL, 2018-02-15:
I have currently some problems with my local network and sometimes it brings up wrong hostnames or other things.

For this it would be helpful to not have to unplug my eps's from energy to restart them without clearing config.

Is there possible yet a wayto do so? Or would it have to be constructed?

timpur, 2018-02-15:
See #458 there is a add already to do this via mqtt and I believe via code (need to double check that )

JorinL, 2018-02-15:
Ah cool!

closed 2018-02-15 by JorinL

timpur, 2018-02-15:
Also homie.reboot() is there

timpur, 2018-02-15:
This release is still a work in progress, but it you like you can use this branch for easily accessible just not it is not been stress tested, but seems stable in my and a couple of others testing. You can also add to this and give feedback if you decide to use it.

JorinL, 2018-02-15:
Thanks for the Feedback :-)

JorinL, 2018-02-25:
One more question -  homie.reboot()  bring @my side 'homie' was not declared in this scope.
Wrong branch :-D ?

I tried it now with:  2.2 and dev. Both seem to be wrong :-D

luebbe, 2018-02-25:
capitalization?

timpur, 2018-02-25:
by bad, Homie.reboot() ???

JorinL, 2018-02-25:
Well simply yes........ Is there something somehow i should have know that? ;-D

Thanks for the help! I really searched for the cause.