homieiot/homie-esp8266 #492: Transfer config.json with success but no normal mode

svenbeb, 2018-02-18:
Hi,

I got a new device (ESP WROM02 with barrery), witch I want to integrate in my mqtt network with homie (like all other ones): Upload Firmaware works fine and without errors.
I Transfered the config.json per curl:
curl -X PUT http://192.168.123.1/config --header "Content-Type: application/json" -d @config.json
and got a success:true

Try it different times, but the device won't go to normal mode.
I can't try the gui mode, because this doesn#t work for me on all devices (wemos d1 mini) but I am happy with the curl method all the time.

The new Device is a:https://www.ebay.co.uk/itm/ESP8266-Esp-Wroom-02-WeMos-D1-Motherboard-Mini-WiFi-Nodemcu-Module-18650-Battery-/292117998785?clk_rvr_id=1444580404404&rmvSB=true

Anybody knows this issue?

Regards
Sven

timpur, 2018-02-19:
Firstly see the doc http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.3/

Secondly see for UI configuration http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.3/advanced-usage/ui-bundle/

Lastly, where the serial out put ? Kind need more info that this.

euphi, 2018-02-19:
Are there any in the custom settings defined in the firmware?
Maybe a mandatory custom setting is missing?

svenbeb, 2018-02-19:
Hi,

thanks for your replies - and I am apologise for the issue.
I read the documentation before and this is not my first homie device - but yesterday it seems that it not want to go to normal mode. Today, I want to follow the hint with the serial (damn, I always forgot this). So, I powered on and it was in normal mode?!
I Don't realy remember if I took out the battery yesterday, to realy reboot/reset the device, maybe this was the fault.

There where no customer settings in the config. And the UI config doesn't work for all of my devices - but as i mentioned, I am fine with the curl method.

SOLVED. THANKS!

closed 2018-02-19 by svenbeb

euphi, 2018-02-20:
Just an additional hint: You need to upload the first software via serial anyhow, so you also can upload the configuration this way.

If you build with platformio, you can also just upload the config with the `platformio run --target uploadfs` command. This builds a SPIFFS from the `data` subdirectory. So you just need to place the configuration in `data/homie/config.json`.

As far as I know the arduino IDE also supports SPIFFS upload.