homieiot/homie-esp8266 #493: Unable to reconnect to wifi

bilogic, 2018-02-19:
Hi,

1. I just started playing around with homie 1.5.0 and managed to get what I needed from it.
2. I also run a dd-wrt router that disable and re-enable its wifi programmatically daily.
3. After the router re-enables its wifi, I can tell from the serial that the homie nodes are stuck in the connecting to wifi phase
4. It only connects properly after I press the esp8266 reset button
5. Option 1: dig really hard to find the issue
6. Option 2: reset the esp8266 programmatically after 10 wifi connection failures

Any other ideas/options? Thank you.

euphi, 2018-02-19:
Option 3: Use Homie 2.0.0 :-)

It is almost stable (and less bugs than 1.5.0).

bilogic, 2018-02-19:
Hmm.. ok, let me give it a shot tomorrow. I tried but failed earlier due to a shallow understanding of MQTT. Thank you.

maxlein, 2018-03-11:
Hi,
I have also the problem of not reconnecting to wifi again. (Wemos D1, Homie 2.0)
The blue led is blinking with a frequency of 1 Hz.

Should I check under Homie.loop() if WIFI is connected and reconnect if not?
If yes, can I somehow read the wifi config of the device for reconnecting?

Or should homie reconnect from itself?

timpur, 2018-03-11:
What version of esp Arduino core ? Homie should reconnect, but sometimes gets stuck and work working to solve these issues.

euphi, 2018-03-11:
I've seen two reasons for this in my projects:

1. sometimes a device does not connect at all after restart.
  --> the cause *may* be a power supply issue, as this happend to me only with devices with a high load on the same power supply (led strips that are switched on).  Most of the time, a reset (reset button) helps. Maybe some internal components of the esp8266ex chip need too long to initialize after power on with low voltage(?).

2. really bad WiFi connection.
Some devices did not reconnect due to bad wifi.
Wifi quality can change fast due to other devices (mobile phones, ...), weather condtions (humidity!) and moving objects (I once placed a metal ladder just in front of a device and wondered why it was offline...).

benzino77, 2018-03-12:
Check what version of esp8266/Arduino framework you are using. I've observed many strange behavior with the newest 2.4.0 version. If you are using platformio you can switch to previous 2.3.x version with this line in `platformio.ini`:

`platform = espressif8266@1.5.0`

At the very beginning of messages, during compilation you should see:
`[03/12/18 07:10:21] Processing esp01 (platform: espressif8266@1.5.0; board: esp01_1m; framework: arduino)`

espressif8266@1.5.0 = esp8266/Arduino 2.3.x
espressif8266@1.6.0 = esp8266/Arduino 2.4.x

maxlein, 2018-03-12:
Thanks for the help!
Will look and try at weekend when I have time!

maxlein, 2018-03-17:
So, I tried version  2.4.1 for a few days now and it worked without problems!
So reconnect works again. Thx for the help.

timpur, 2018-03-18:
Keep an eye on it and get back to use if there are any issues, would be nice if it is fixed

closed 2019-03-30 by stritti