homieiot/homie-esp8266 #494: MQTT topic homie/device_id/$nodes missing?

jowi24, 2018-02-19:
I created a simple homie application measuring values from a temperature sensor. All is working fine except there is a missing mqtt topic, which should be there if I read homie convention docs correctly:

homie/device_id/$nodes

I'm using v2.0.0-beta.3. Am I expected to do something special in my code to get this topic? I assumed it gets auto-generated based on the nodes I instantiated.

jowi24, 2018-02-19:
This is a stripped down version of my code:

```main.cpp
#include <Homie.h>
#include <DHT.h>

const int SENSOR_INTERVAL = 60;

unsigned long lastSensorvaluesSent = 0;

DHT dht(14, DHT11);

HomieNode temperatureNode("temperature", "temperature");
HomieNode humidityNode("humidity", "humidity");

void setupHandler() {
  temperatureNode.setProperty("unit").send("°C");
  humidityNode.setProperty("unit").send("%");
}

void loopHandler() {
  if (millis() - lastSensorvaluesSent >= SENSOR_INTERVAL * 1000UL || lastSensorvaluesSent == 0) {
    float temperature = dht.readTemperature(); 
    Homie.getLogger() << "Temperature: " << temperature << " °C" << endl;
    temperatureNode.setProperty("degrees").send(String(temperature));

    float humidity = dht.readHumidity();
    Homie.getLogger() << "Humidity: " << humidity << " %" << endl;
    humidityNode.setProperty("value").send(String(humidity));

    lastSensorvaluesSent = millis();
  }
}

void setup() {
  Serial.begin(9600);
  Serial << endl << endl;

  Homie.setLedPin(2, LOW);
  Homie_setFirmware("THB-Sensor", "0.0.3");
  Homie.setSetupFunction(setupHandler).setLoopFunction(loopHandler);
  
  temperatureNode.advertise("unit");
  temperatureNode.advertise("degrees");
  humidityNode.advertise("unit");
  humidityNode.advertise("value");

  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

timpur, 2018-02-19:
Homie ESP 2.0 uses Homie 2.0 see homie convention releases to get the specs of that version

jowi24, 2018-02-20:
Okay, I see, this is an extension introduced in version 2.1.0. Thank you for the clarification!

closed 2018-02-20 by jowi24