homieiot/homie-esp8266 #495: Fail to reboot esp8266

erow, 2018-02-23:
When I updated configuration by mqtt, and it succeed. But it didn't reboot normally, I got this:
```
(Temperature °C,Humidity %): 11.11,66.35
✔ Configuration updated
Flagged for reboot
Device is idle
↻ Rebooting...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...

Exception (28):
epc1=0x4000df2f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000033 depc=0x00000000

ctx: sys
sp: 3ffffa90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffc30:  3ffefde6 3fff411c 402593b0 401066e0
3ffffc40:  4022dcab 00000006 00000005 4023c14d
3ffffc50:  00000033 00000010 4022e621 3ffef120
3ffffc60:  3fff411c 00000000 3ffe9766 3fff29e4
3ffffc70:  3fff411c 4022e549 3ffef120 40232628
3ffffc80:  3ffe9766 00000002 0000001a 00000000
3ffffc90:  3fff421b 3fff273c 00000005 3ffef874
3ffffca0:  3ffe9766 3ffe9766 00000033 4023052f
3ffffcb0:  3ffef828 3ffef920 40232628 3ffef828
3ffffcc0:  402515cd 402515d0 3ffef1a4 40212641
3ffffcd0:  00000001 3ffee9f0 00000011 40220e3e
3ffffce0:  4023014f 40230120 3ffee9f0 40212730
3ffffcf0:  40231fe0 40230120 4023a8eb 00000000
3ffffd00:  3ffef1a4 40231fa2 3ffef120 00000012
3ffffd10:  40230116 00040000 7fffffff 00000000
3ffffd20:  402300bf 3ffef120 00000012 40232628
3ffffd30:  40230628 3ffef120 00000000 40232628
3ffffd40:  4023009d 4023004e 00000002 00000000
3ffffd50:  402323e1 0000010f 3fff03f8 00000000
3ffffd60:  00000004 0000010f 3ffef348 40232292
3ffffd70:  4023229e 3fff0419 3fff03f8 00000001
3ffffd80:  40242d30 3fff0419 3fff03f8 00000000
3ffffd90:  3fff273c 40215a89 3fff0419 40215997
3ffffda0:  41414141 ff070041 0000e23f 00000300
3ffffdb0:  ff16df00 2204543f ff313c40 0000003f
3ffffdc0:  31383538 37363930 ff1a4c00 0000193f
3ffffdd0:  00001900 ff038000 ff2c303f ff20683f
3ffffde0:  ff00f53f 1004d83f 24041240 ff00f540
3ffffdf0:  ff20683f 106cc83f 24174940 0002cb40
3ffffe00:  3e109400 41c51696 41414141 007f8000
3ffffe10:  0000007e 6000001c ff000000 70706100
3ffffe20:  00000000 38ff1bc4 30313835 00373639
3ffffe30:  3fff1da4 00000019 00000019 3ffeee60
3ffffe40:  3fffff40 3fff23d0 3fff03d5 40222198
3ffffe50:  40241c7e 3fff03d5 3fff23d0 401066e0
3ffffe60:  40242fb1 00ff4398 3fff3e0c 40242f96
3ffffe70:  00000000 3fff23d0 00000001 00000000
3ffffe80:  4021525c 00000030 3fff037c 3ffeee60
3ffffe90:  3fff03f8 3fff23d0 00000000 4020b4e6
3ffffea0:  401045e4 06e33ee3 3fff2d6c 00000000
3ffffeb0:  3ffeecc0 00000001 3ffe9ea7 402212b0
3ffffec0:  00000001 06de7a2f 4021e43c 3ffeee60
3ffffed0:  3fffff40 3fff037c 3fff037c 40220e5c
3ffffee0:  40208440 3fffff58 3fffff58 40212641
3ffffef0:  3ffe9ea6 3fff037c 00000024 40220e3e
3fffff00:  3fff0870 3f800000 3fff0180 3ffeee60
3fffff10:  3fffff40 3fff0870 3fff0180 4020c011
3fffff20:  3fff23d4 3fff3d1c 3fff40ec 40220c63
3fffff30:  000005e0 00000000 3fff40ec 402151b0
3fffff40:  3fff4404 0000000f 00000005 963e1094
3fffff50:  4010c516 00000008 00000000 00000000
3fffff60:  00000000 06de84b8 3fff09c0 40210026
3fffff70:  402403d8 3fff40ec 3fff3d44 402153e6
3fffff80:  402403fe 3fffdab0 00000000 3ffeee60
3fffff90:  3fffdc10 00000000 3fff40ec 402445d0
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
If I reset esp by button, everything is OK. I don't know what's wrong. It seams Homie call `ESP.restart();` simply. Maybe it's a problem of ESP library.

timpur, 2018-02-25:
Is this just after flashing? Does that happen in every time ?

erow, 2018-02-25:
you mean I calling reboot directly?

timpur, 2018-02-25:
I mean is this the first reboot of the device after flashing firmware ?

Also there is Homie.reboot()

erow, 2018-02-26:
Yes.
And I call `ESP.restart()` in loop, it can reboot successfully.
```
Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.2.4
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Calling setup function...
〽 Sending statistics...
  • Wi-Fi signal quality: 86%
  • Uptime: 3s
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...

Exception (28):
epc1=0x4000df2f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000033 depc=0x00000000

ctx: sys
sp: 3ffffa90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffc30:  3fff00e6 3fff3234 4025a4f0 401066e0
3ffffc40:  4022ebd3 00000006 00000005 4023d079
3ffffc50:  00000033 00000010 4022f549 3ffef420
3ffffc60:  3fff3234 00000000 3ffe9902 3fff311c
3ffffc70:  3fff3234 4022f471 3ffef420 40233550
3ffffc80:  3ffe9902 00000002 0000000e 00000000
3ffffc90:  3fff32bf 3fff2e74 00000009 3ffefb74
3ffffca0:  3ffe9902 3ffe9902 00000033 40231457
3ffffcb0:  3ffefb28 3ffefc20 40233550 3ffefb28
3ffffcc0:  402524f9 402524fc 3ffef4a4 3fff4b92
3ffffcd0:  00000001 4023a933 3ffee2b8 3fff4a14
3ffffce0:  40231077 40231048 00000005 402212cf
3ffffcf0:  40232f08 40231048 4023b813 00000001
3ffffd00:  3ffef4a4 40232eca 3ffef420 00000012
3ffffd10:  4023103e 00000000 3fff2e74 000000f6
3ffffd20:  40230fe7 3ffef420 00000012 40233550
3ffffd30:  40231550 3ffef420 00000000 40233550
3ffffd40:  40230fc5 40230f76 00000002 00000000
3ffffd50:  40233309 0000010f 3fff0b50 00000000
3ffffd60:  00000004 0000010f 3ffef648 402331ba
3ffffd70:  402331c6 3fff0b71 3fff0b50 00000001
3ffffd80:  40243c60 3fff0b71 3fff0b50 00000000
3ffffd90:  3fff2e74 402168c5 3fff0b71 402167d3
3ffffda0:  70736152 72726562 00000079 00000300
3ffffdb0:  ff20df00 2222643f ff3bdc40 0000003f
3ffffdc0:  34333231 38373635 ff247c00 0000193f
3ffffdd0:  00001900 ff0dd800 ff36d03f ff2aa83f
3ffffde0:  ff0b2d3f 2231603f 242bae40 ff0b2d40
3ffffdf0:  ff2aa83f 1066e03f 243ee140 0002dc40
3ffffe00:  78100800 52cd7899 62707361 79727265
3ffffe10:  00000000 6000001c ff000000 00000a0d
3ffffe20:  00000000 31ff229c 35343332 00383736
3ffffe30:  3fff247c 00000019 00000019 3ffef158
3ffffe40:  3fffff40 3fff2aa8 3fff0b2d 40223160
3ffffe50:  40242bae 3fff0b2d 3fff2aa8 401066e0
3ffffe60:  40243ee1 00000006 3fff4c64 40243ec6
3ffffe70:  00000000 3fff2aa8 00000001 00000000
3ffffe80:  40216098 3fff4a18 3fff0ad4 3ffef158
3ffffe90:  3fff0b50 3fff2aa8 00000000 4020ca6a
3ffffea0:  401045e4 0015d45e 3fff4864 00000000
3ffffeb0:  3ffeefc0 00000001 3ffea5b5 40222264
3ffffec0:  00000001 00110faa 40220b6c 3ffef158
3ffffed0:  3fffff40 3fff0ad4 3fff0ad4 40221ed8
3ffffee0:  40209d68 3fffff58 3fffff58 4021344d
3ffffef0:  3ffea5b4 3fff0ad4 00000024 40221eba
3fffff00:  3fff0fc8 3f800000 3fff08c8 3ffef158
3fffff10:  3fffff40 3fff0fc8 3fff08c8 4020d6a9
3fffff20:  3fff2aac 3fff447c 3fff45c4 40221cd7
3fffff30:  000005e0 00000000 3fff45c4 40215fec
3fffff40:  3fff45f4 0000000f 00000009 99781008
3fffff50:  0000cd78 00000008 00000000 00000000
3fffff60:  00000000 001114ad 3ffeefc0 00370cd4
3fffff70:  40241308 3fff45c4 3fff44a4 40216222
3fffff80:  4024132e 3fffdab0 00000000 3ffef158
3fffff90:  3fffdc10 00000000 3fff45c4 40245500
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
```

timpur, 2018-02-26:
If it is the first reboot after flashing then this can be ignored. It has been well known that some weird things can happen the first time after flashing. If it happens over many reboots after a single flash then this is an issue.

closed 2018-02-26 by timpur