homieiot/homie-esp8266 #499: Unable to flah via OTA in certain condition

nos86, 2018-02-28:
Hi,
first of all I'd like to say really thank you for your work.. It made my life easier.

I'm writing you to report a very strange behavior I saw using OTA withHomie-OTA server and Homie v2.2.
In particular, I spent all this afternoon trying to understand why I wasn't able to flash anymore.
At the end, I don't know the reason, but I identified the conditions where I see the problem and how to avoid it.

Starting from the issue, here what ESP8266 is saying:

> Receiving OTA payload
> ↕ OTA started
> Triggering OTA_STARTED event...
> Firmware is base64-encoded
> Receiving OTA firmware (1035/407073)...
> Receiving OTA firmware (2130/407073)...
...................................................................................
> Receiving OTA firmware (403803/407073)...
> Receiving OTA firmware (404898/407073)...
> Receiving OTA firmware (405993/**407073**)...
> Receiving OTA firmware (**407071/407072**)...
> ✖ OTA failed (400 BAD_FIRMWARE)
> Triggering OTA_FAILED event..

After some investigation, I understood the problem is the SIZE_ERROR; in fact, as you can see the total size at the last row is different than the others (407072 != 407073) and the downloaded bytes are less than the total (407071/407072).
Attached you can find the bin file where I have the problem: 1.0.4 is the current one and 1.0.3 is new firmware I want to flash.

After several trials, I tried to switch the encoding option from base64 to binary and I fixed ( or maybe is better to say I avoided) the issue.

For the moment, I solved my problem, anyway, I want to highlight this case as a possible bug.

Thanks again for your work,
Salvo

[Uploading bare-minimum.zip…]()

timpur, 2018-02-28:
Thank you for all the research. Something to fix when I get the time :)

Gulaschcowboy, 2018-02-28:
Hi @timpur sounds related to my issue. So when you have time, I can offer testing/debug resources. Thank you!

mverleun, 2018-04-12:
I'm experiencing a similar problem. But the problem is not persistent.
After several attempts to upload the firmware it will successfully upload.
When the uploads fail I'm getting the 'bad firmware' message.
Before the successful upload the ota status of the device eventually becomes 500, after which I reset the Wemos. The subsequent upload then suddenly works fine.