homieiot/homie-esp8266 #500: can't compile sketch

kdhl, 2018-03-09:
I installed all libraries according to the 'getting started' and they show up in my 1.8.3 arduino IDE. I'm using a wemos d1 mini which works when I load the 'blink'.
But when i try to compile the 'bare minimum sketch' I get the error: 'endl' was not declared in this scope. Other examples bring different errors, 'TemperatureSensor' brings: Error compiling for board WeMos D1 D2 & mini, same error on 'LightOnOff' etc

Full error message:
_Arduino: 1.8.3 (Windows 7), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

d:\Users\bert\Documents\Arduino\libraries\homie-esp8266-1.5.0\src\Homie\MqttClient.cpp: In member function 'void HomieInternals::MqttClient::initMqtt(bool)':

d:\Users\bert\Documents\Arduino\libraries\homie-esp8266-1.5.0\src\Homie\MqttClient.cpp:32:143: error: no matching function for call to 'PubSubClient::setCallback(std::_Bind_helper<false, void (HomieInternals::MqttClient::*)(char*, unsigned char*, unsigned int), HomieInternals::MqttClient* const, const std::_Placeholder<1>&, const std::_Placeholder<2>&, const std::_Placeholder<3>&>::type)'

   this->_pubSubClient.setCallback(std::bind(&MqttClient::_callback, this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3));

                                                                                                                                               ^

d:\Users\bert\Documents\Arduino\libraries\homie-esp8266-1.5.0\src\Homie\MqttClient.cpp:32:143: note: candidate is:

In file included from d:\Users\bert\Documents\Arduino\libraries\homie-esp8266-1.5.0\src\Homie\MqttClient.hpp:4:0,

                 from d:\Users\bert\Documents\Arduino\libraries\homie-esp8266-1.5.0\src\Homie\MqttClient.cpp:1:

d:\Users\bert\Documents\Arduino\libraries\ESP8266_Microgear/PubSubClient.h:115:18: note: PubSubClient& PubSubClient::setCallback(void (*)(char*, uint8_t*, unsigned int))

    PubSubClient& setCallback(MQTT_CALLBACK_SIGNATURE);

                  ^

d:\Users\bert\Documents\Arduino\libraries\ESP8266_Microgear/PubSubClient.h:115:18: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (HomieInternals::MqttClient::*)(char*, unsigned char*, unsigned int), HomieInternals::MqttClient* const, const std::_Placeholder<1>&, const std::_Placeholder<2>&, const std::_Placeholder<3>&>::type {aka std::_Bind<std::_Mem_fn<void (HomieInternals::MqttClient::*)(char*, unsigned char*, unsigned int)>(HomieInternals::MqttClient*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>}' to 'void (*)(char*, uint8_t*, unsigned int) {aka void (*)(char*, unsigned char*, unsigned int)}'

exit status 1
Error compiling for board WeMos D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
_

No doubt I'm doing something wrong, but what? Thanks in advance for your feedback.

njerig, 2018-03-09:
Both 'endl' and 'Homie_setFirmware' are not declared in the scope for me. I'm on MacOS 10.13.3 though, using PlatformIO Core 3.5.2.

Full error msg:

```
[Fri Mar  9 15:43:14 2018] Processing d1_mini (platform: espressif8266; lib_deps: Homie; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 36 compatible libraries
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <Homie> v1.5.0
|   |-- <ArduinoJson> v5.13.1
|   |-- <PubSubClient> v2.6
|   |-- <Bounce2> v2.41
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <Ticker> v1.0
|   |-- <ESP8266WebServer> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <DNSServer> v1.1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266httpUpdate> v1.1
|   |   |-- <ESP8266HTTPClient> v1.1
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
Compiling .pioenvs/d1_mini/src/main.cpp.o
Compiling .pioenvs/d1_mini/FrameworkArduino/IPAddress.cpp.o
Compiling .pioenvs/d1_mini/FrameworkArduino/MD5Builder.cpp.o
Compiling .pioenvs/d1_mini/FrameworkArduino/Print.cpp.o
Compiling .pioenvs/d1_mini/FrameworkArduino/Schedule.cpp.o
Compiling .pioenvs/d1_mini/FrameworkArduino/Stream.cpp.o
Compiling .pioenvs/d1_mini/FrameworkArduino/StreamString.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:33:13: error: 'endl' was not declared in this scope
Serial << endl << endl;
^
src/main.cpp:35:44: error: 'Homie_setFirmware' was not declared in this scope
Homie_setFirmware("bare-minimum", "1.0.0");
^
*** [.pioenvs/d1_mini/src/main.cpp.o] Error 1
```

kdhl, 2018-03-11:
I solved it by using the homie-esp8266-2.0.0-beta.3 library. First I had 1.5.0 since that was marked as 'stable'.
now sketches compile without problem; compiling only mentions: "WARNING: Spurious .circleci folder in 'Homie' library" but that don't seem to be a problem
thanks for writing homie!

closed 2018-03-11 by kdhl

euphi, 2018-03-11:
@marvinroger , @timpur you should release 2.0.0 on platformio to avoid issues like this. 

Any bug fixes etc. before 2.1 can be released as 2.0.1 etc.

timpur, 2018-03-12:
@euphi look at #454- maybe you can help me make relase notes