homieiot/homie-esp8266 #508: How to have multiple loops?

psamim, 2018-03-27:
Hi,

I am using Homie 2. I want the button to also work in offline mode. The Homie loop is called only when Homie is connected. How can I have another loop? 

Should I use TaskScheduler and run Homie on another task? Because Homie setup blocks the setup function and if there is no wifi it cannot reach the loop function.

Thanks

amayii0, 2018-03-27:
Probably related to #450

euphi, 2018-03-27:
See also #487, #257.

I created an PR (#490) that also calls the loopHandler when Homie is disconnected.

euphi, 2018-03-27:
You may want to try this PR, just  download the patch (https://github.com/marvinroger/homie-esp8266/pull/490.patch) and apply it to your local Homie sources.

psamim, 2018-03-27:
Thanks! I used TaskScheduler and it is working OK.

closed 2018-03-27 by psamim