homieiot/homie-esp8266 #509: Questions regarding SPIFFS

psamim, 2018-03-27:
Hi!

I am new to Homie and Platformio. I read the documents and what I do for adding the `config.json` file is running:
- `platformio run --target uploadfs`
- `platformio run --target upload`

Is the order correct? Does it matter? Do I have to do them both? I am asking this because sometimes even after doing the `uploadfs` I get the error `config.json doesn't exist`, but sometimes it works. Sometimes everything is working but suddenly it reboots and the configuration is gone/erased. So I thought maybe I am doing something wrong.

Thanks

psamim, 2018-03-27:
This is the log:

```
Flagged for reset by pin

Device is idle
Configuration erased
Setting next boot mode to 2
Triggering ABOUT_TO_RESET event...
↻ Rebooting into config mode...
✖️ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
✖️ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Is some pin reserved for config erase?

euphi, 2018-03-27:
Yes, pressing Pin 0 (named D3 on Wemos D1 Mini) for 5 seconds deletes the configuration.

You can disable this function with (see http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.3/advanced-usage/resetting/)

```C++
void setup() {
  Homie.disableResetTrigger(); // before Homie.setup()
  // ...
}
```

psamim, 2018-03-27:
Thanks! And does the order matter? Is the way I put config.json correct?

euphi, 2018-03-28:
The order doesn't matter, so the way you upload it is correct.

However, please be careful with the flash size, because this has impact on the layout ("partitioning") of the program segment and SPIFFS.

For some boards (ESP-07, Wemos D1 Mini) the actual flash size differs from what is pre-configured in platformio, so you need to configure it manually with `build_flags` in platformio.ini. See http://docs.platformio.org/en/latest/platforms/espressif8266.html#custom-flash-size

If configured flash size does not match the actual flash size you may be stuck in a reset loop or the `config.json` cannot be found (I think, what happens then is related to flashing the application binary or the SPIFFS image first....)

psamim, 2018-04-03:
@euphi Thanks for the insights.

closed 2018-04-03 by psamim