homieiot/homie-esp8266 #510: config.json lost after power off

mosseb, 2018-03-28:
Hi.
I have my first little homie project, which I upload on a nodemcu v2 using platformio and atom.
I upload the project and upload the spiffs correctly.
I have a data/homie/config.json written properly : homie boots it normal mode.
When I hit reset, it restarts in normal mode as expected.
When I unplug USB and plug it again, homie starts in config mode (permanent red led). Serial says "Configuration invalid. Using CONFIG MODE".
If I publish again the spiffs, it works again.
At my friend's home, which has exactly the same hardware, and the same code, it works normally.
I tried on several nodemcus, fearing a hardware problem.

Do you have any advice ?
Thanks.
Regards,
mosseb

mosseb, 2018-03-28:
I investigated a bit more and it does not seem homie related at all...
If I plug it in wall charger directly, it runs OK.
If I plug it on my PC, the config is lost, even if I reset it, it's irremediably lost.
If I remove everything USB from my pc and try again, doen't work neither.
If I mesure the voltage of the nodemcu plugged into wall charger, I get 4.55V.
If I mesure the same plugged into my PC I get 4.59V.
This does not seem to be a problem.
I had problems with reading some usb key on this pc, but I cannot figure out how this is related, because I'm able to program it without problem once it's plugged...
I'm waiting for some feedback before closing this issue because it does not seem homie related but I would be interested in your opinion.
Regards.

mosseb, 2018-03-31:
To disambiguate the problem I bought a pciexpress<=>USB3 card.
This is working perfectly now.
So that means my PC USB is crap, might be some problem with power supply or motherboard.
Anyway i'ts working now I can peacefully continue to work on my shutters :)
Sorry for polluting issue tracker.

closed 2018-03-31 by mosseb

timpur, 2018-04-01:
Glad you solved it :)