homieiot/homie-esp8266 #511: 'class HomieInternals::HomieClass' has no member named 'setNodeProperty'

JorinL, 2018-03-31:
Hey there,

I'm not sure if this is coming by using a wrong version (tested with develop,master,release 2.0.0) or if I just use it wrong.

I have a node which reacts on "reboot" - but to avoid a reboot loop it should publish after receiving that payload into my topic "restarting" so it won't reboot over and over....

I tried it first with .send("restarting") which does not avoid the reboot loop.
Now with " Homie.setNodeProperty" it won't compile.

Where is my error?

`  else if (command == "reboot") {
    //roombaNode.setProperty("command").send("restarting");
    Homie.setNodeProperty(roombaNode, "command", "restarting");
    Homie.reboot();
  }`

JorinL, 2018-03-31:
Damn.... that was easy... got an simple idea an it was the solution. :-D

`else if (command == "reboot") { 
roombaNode.setProperty("command/set").send("restarting"); 
Homie.reboot();
}`

timpur, 2018-03-31:
What you should do is receive the command 'restart' though the set handler like normal. When reacting to this command you should set the prop command to 'restarting' and set the /set prop to '' (blank). Thus when you restart there is on /set to be received.

JorinL, 2018-03-31:
Very nice tip! Thanks!

timpur, 2018-03-31:
if that made sense :P

JorinL, 2018-03-31:
Yeah it really helped me.
I just wanted to not step into a loop by using reatin on MQTT.
So I tried to set up something that chaged the payload of the topic and so prohibit a loop.
Setting it to blank is the nicest way :-)

timpur, 2018-04-01:
I believe sending a null retained messages to a topic removes the retained message. So the broker actually won't send out anything when first subscribing. (If I understand mqtt specs)

JorinL, 2018-04-02:
Got a confusing situation:

I own various SonOff Basic Switches and SonOff S20 and integrated to all of both types the reboot and reset with setting payload to null viar

JorinL, 2018-04-02:
Mhh i found something:

> sonoff-basic-1/$homie 2.0.1
sonoff-s20-1/$homie 2.0.0
sonoff-s20-2/$homie 2.0.0
sonoff-s20-3/$homie 2.0.0
sonoff-s20-4/$homie 2.0.0

euphi, 2018-04-02:
Just one comment regarding the original post: Why do you need to send the reboot command as "sustained" message? There is usually no need to send commands sustained.

JorinL, 2018-04-03:
In general it can help you, if you have devices which sleep long time and work on battery and only wake up triggered by sensor events. Then you say the device to reset/reboot/stay online/update next time it goes online and does its job.

Also I like to have it traceable and predictable - so sustained payloads help me keeping the overview.

But for this part here now its only the question why the versions don't match - since I used the same library for updating them and if that is the cause for not running into that if condition.

JorinL, 2018-04-04:
Issue solve - my fault... had developement in use...  🥇

closed 2018-04-04 by JorinL