homieiot/homie-esp8266 #513: Wifi Reconnect Issue

psamim, 2018-04-03:
Hi,

I am using development version of Homie on Wemos mini d1 using Platformio.

I works flawlessly but I have one problem. If the device is turned on and there is no wifi and the wifi becomes available later, it will not connect to wifi. It can connect to the same wifi if it is available on the first try. It also cannot reconnect to wifi if it becomes unavailable.

There is another related strange issue. If I turn the device off while it is disconnected from wifi, the program is erased and it is reset.

Thanks

psamim, 2018-04-03:
I tested the same scenario with mqtt. I stopped my mqtt server and started it again. The device correctly reconnects to mqtt but wifi reconnect does not work.

stritti, 2020-12-28:
Could you retest this issue with latest version, please?
If it is still an issue, please let us know.

psamim, 2020-12-28:
@stritti Thanks but unfortunately I'm not using Homie anymore and I do not have the devices to test again, sorry (it was two years ago).

stritti, 2020-12-28:
okay, then we close this issue.

closed 2020-12-28 by stritti