homieiot/homie-esp8266 #516: Troubleshooting "Attempting to connect to MQTT Server"

RumHamRadio, 2018-04-13:
I've tried setting up Homie-ESP8266 v 2.0 in Arduino IDE (with two different NodeMCU modules to rule out hardware issues), but I'm never able to get past a neverending loop of "Attempting to connect to MQTT Server" in the Arduino IDE serial monitor.

My Mosquitto server is running on my development machine (Windows computer) and I've confirmed that it is running properly using mosquitto_sub and mosquitto_pub. I collected my server's local IPv4 address using ipconfig and used that in the Homie configurator, but the 'Attempting to connect' loop continues.

What is the best way to troubleshoot why my ESP8266 modules are unable to connect to my MQTT server? Here's the info from the serial monitor, in case that's needed:

💡 Firmware bare-minimum (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: 68c63ac36aa1
  • Device ID: testdevice
  • Name: TestDevice
  • Device Stats Interval: 60 sec
  • Wi-Fi: 
    ◦ SSID: SHOP
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 192.168.20.84
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  • OTA: 
    ◦ Enabled? no
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.20.81
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...

euphi, 2018-04-15:
Do you have access to the MQTT server? Try to run tcpdump/ethercap/wireshark to capture TCP traffic.

RumHamRadio, 2018-04-16:
Thanks for the reply! You set me in the right direction (Wireshark wasn't picking up any MQTT protocol messages) and I figured out what the issue was.

I was using Windows Services to manage Mosquitto's services, which didn't ask me to set an exception for Mosquitto in Windows Firewall. Here's how I fixed it:

1. In Windows, search 'Services' and open the View Local Services dialog (Or use Windows Key+R, then in the Run dialog, type services.msc and hit enter to open the View Local Services dialog)
2. Scroll down to Mosquitto Broker in the list of services, right-click it, and choose to STOP the service
3. Open a CMD prompt, and type `cd c:\Program Files (x86)\mosquitto\` and hit enter to navigate to the Mosquitto install directory
4. Type `mosquitto -v` and hit enter
5. A Windows Firewall Exception dialog box will pop up. Choose to allow Mosquitto to communicate on your network in this dialog box.

The Mosquitto broker is now able to communicate on the network.

closed 2018-04-16 by RumHamRadio