homieiot/homie-esp8266 #518: Looking for help

timpur, 2018-04-17:
Wondering if anyone wants to contribute back to homie. Looking for help with C++ homie esp8266 but also some react dev (web dev) for new homie web ui.

luebbe, 2018-04-18:
I can try to help, but I'm not really inside the homie framework, just a user from the outside.

amayii0, 2018-04-18:
Also interested but not even a user yet, just keeping curious.
My use cases would be to:

1. Manage Mitsubishi HVAC appliances over IR (or maybe serial) : I have control working but no real frontend (looking into Home Assistant and climate control but not very happy with it ATM)
2. Run a 24h clock using a stepper motor : Very early dev, will require NTP for init and including a I2C RTC. Motor does work fine but not yet using MQTT in any way

Will keep curious about Homie anyway and create Instructables when time permits (see https://www.instructables.com/member/RudyD1/instructables/ )

timpur, 2018-04-18:
Thanks guys

LennartHennigs, 2018-05-09:
Hey, I am a homie user and my C++ is still a bit rusty (back from university)  but I have started to code again over the last two years. So yes, I can try to lend a hand.

timpur, 2018-05-09:
Currently im working on:
* homie-esp8266 v2.1 ([c++](https://github.com/marvinroger/homie-esp8266/pull/458))
* homie-config v2.0 (react (preact) & ts) (no repo yet)
* homie-esp32 v1.0 ([c++](https://github.com/timpur/homie-esp8266/tree/esp32))
* homie discovery for python v1.0 ([python](https://github.com/timpur/homie-discovery-python))
* Home Assistant addon for Homie v1.0 ([python](https://github.com/nerdfirefighter/HA_Homie))
* Arduino Async Programming Style v1.0 (c++) (not yet started, [research](https://github.com/LifeWanted/liblw/issues/3#issuecomment-387579200))

@marvinroger @luebbe @amayii0 @LennartHennigs @nerdfirefighter @mrpace2 @euphi @ivankravets @bertmelis @ThomDietrich @tripflex @gulaschcowboy @flaviostutz 

Sorry for spam, just wondering if anyone is interested in these. Im doing this after work and find it hard to get time to do all this :P ....

marvinroger, 2018-05-09:
Hi guys,

I would like to thank you all for all the work done here. I am proud of the project, but most importantly proud of the awesome community that we are. What started as a small personal project ended up being an almost 1k starred repo, with many IoT projects interested in the Homie convention, and this could not have happened without the help of every single individual who participated.

The thing is, over time, I had less and less free time to work on OSS projects. I am not just a student anymore, I now have to manage work too, and with family and friends, school, 100+ GitHub emails per day, OSS code, shaping up my professional project, and so on, I unfortunately don't really have time for the project, as you probably noticed.

Awesome folks, like @timpur, allowed the project to stay alive, so thanks for that. 😉

paumito, 2018-05-25:
I'm interested to help!

timpur, 2018-05-25:
@paumito, which project :P see above ?
ps, thanks :)

s00500, 2018-05-28:
My time is also limited but maybe I can help too, I could help with the react project and also all esp related code

timpur, 2018-05-28:
Find me on discord `timpur#5823`

Thanks guys :)

timpur, 2018-05-28:
FYI: updated my list ^

paumito, 2018-05-29:
Woah haha just forgot to choose the project :P 

esp-32(c++) seems good. Any projects with C++ is ok for me.

closed 2019-07-15 by stritti