homieiot/homie-esp8266 #520: Question: How to properly set node property attributes

mwildbolz, 2018-05-01:
Hi!

What's the intended way to set all node property attributes according to homie convention?
How would I implement e.g. the example from the convention resulting in
```
homie/super-car/engine/temperature/$name → "Engine temperature"
homie/super-car/engine/temperature/$settable → "false"
homie/super-car/engine/temperature/$unit → "°C"
homie/super-car/engine/temperature/$datatype → "float"
homie/super-car/engine/temperature/$format → "-20:120"
```

timpur, 2018-05-01:
Homie for esp8266 is using version 2.0.1 of the convention. The above is a newer 3.0 version of the convention. Are you trying to implement 3.0 ? This will be a huge rework and homie esp8266 3.0 has been discussed in another tab to have more than just the update convention.

mwildbolz, 2018-05-06:
Sorry, wasn't clear for me that this is already Homie 3.0
At the moment I'm not working on a rework, was just a question how to handle this with the current API...

closed 2018-05-06 by mwildbolz

timpur, 2018-05-06:
It does say in the readme I believe (latest)