homieiot/homie-esp8266 #521: $stats/interval always 0

mwildbolz, 2018-05-01:
The topic $stats/interval is always 0.
Maybe because in L359 there is a division by 1000 where the value is already in seconds - for me there is no need to do this division.

https://github.com/marvinroger/homie-esp8266/blob/99161ce03c7ff1c8d3dc1fcc52abbf081212b68f/src/Homie/Boot/BootNormal.cpp#L359

timpur, 2018-05-02:
why is there no need for 'you' ? are you changing the $interval value from the default?

mwildbolz, 2018-05-06:
No need, because regarding to Homie specification this value is in seconds and the STATS_SEND_INTERVAL_SEC is also a value in seconds. So you don't need to divide it by 1000 - would only be necessary if the value is in ms.

And yes, I'm also changing the device_stats_interval through configuration.
I would also expect, that the changed value is written to this Homie topic.

CapnBry, 2018-06-10:
This report is correct, STATS_SEND_INTERVAL_SEC is in seconds and the code divides it by 1000 expecting it to be in milliseconds before outputting it. In the code:
```
  const uint32_t STATS_SEND_INTERVAL_SEC = 1 * 60;
```
Where it uses the interval it also expects seconds:
```
  _statsTimer.setInterval(Interface::get().getConfig().get().deviceStatsInterval * 1000);
```
So one of the two is wrong and I believe @mwildbolz is correct considering the constant name is SEC and not MSEC.

timpur, 2018-06-11:
Easy to fix, will do. Thanks guys

stritti, 2019-04-04:
@timpur could you confirm that it is corrected already?

kleini, 2019-04-04:
I can confirm, that $stat/interval is still zero for all of my devices using actual develop branch.

tantive, 2019-04-26:
If I remember corrected I fixed that one when I ported the lib to esp32 and made it play nicely with openHab (which requires repulishing interval in order to show a node as online).
So it should be fixed already in the develop-v3 branch.

stritti, 2019-08-02:
Works for me in combination of develop-V3 and openHab

closed 2019-08-02 by stritti