homieiot/homie-esp8266 #522: Cannot change config by publishing to $implementation/config/set on ESP-01

jmozmoz, 2018-05-19:
Using the bare minimum example, I am trying to update the settings using the way described at https://marvinroger.github.io/homie-esp8266/docs/develop/others/ota-configuration-updates/

This is the command to publish the new config and the json file:
`mosquitto_pub -h alpha  -p 1883   -u xxx -P 'yyy' -t 'homie/bare/$implementation/config/set' -f test.json`

```
{
        "name": "bare_up"
}
```

This is the output on the serial console:
```
✔ Configuration updated
Flagged for reboot
Device is idle
↻ Rebooting...

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

The output from the broker during the update process:
```
homie/bare/$implementation/version 2.0.0
homie/bare/$implementation/ota/enabled true
homie/bare/$online true
homie/bare/$implementation/config/set {
        "name": "bare_up"
}

homie/bare/$implementation/config/set (null)
homie/bare/$online false
```

Then the ESP-01 stops.

If I reset it manually, the output on the serial device is:
```
Configuration invalid. Using CONFIG MODE
💡 Firmware bare-minimum (1.0.0)
🔌 Booting into config mode 🔌
Device ID is xxxxxxxxx
AP started as Homie-xxxxxxxxxx with IP 192.168.123.1
Triggering Wi-Fi scan...
✔ Wi-Fi scan completed
```

Here is main.cpp and platformio.ini:
```
#include <Homie.h>

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;
  Homie.disableLedFeedback();

  Homie_setFirmware("bare-minimum", "1.0.0"); // The underscore is not a typo! See Magic bytes
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
lib_deps = https://github.com/marvinroger/homie-esp8266.git

; Serial Monitor options
monitor_speed = 115200
```

timpur, 2018-05-19:
Think this was a bug, but has been fixed in my fork V2.1, if you have a sec, could you test on my v2.1 ?

jmozmoz, 2018-05-19:
Unfortunately, this does not solve the problem:
```
💾 Saved config file.
✔ Configuration updated
Flagged for reboot
Device is idle
Triggering ABOUT_TO_RESTART event...
↻ Rebooting...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

↻ Config file loading
✖ Config file faild to load
✖ /homie/config.json doesn't exist
Configuration invalid. Using CONFIG MODE
💡 Firmware bare-minimum (1.0.0)
🔌 Booting into config mode 🔌
Device ID is xxxxx
AP started as Homie-xxxxx with IP 192.168.123.1
```

timpur, 2018-05-19:
Odd, I'll have a look into later today

jmozmoz, 2018-05-27:
Did you find any reason for this effect/bug? Can I help debugging?

timpur, 2018-05-27:
Sorry, lacking for time... Will try got to this asap

plambrechtsen, 2018-06-04:
Are you using an ESP-01 with a PUYA flash on it?? I ran into exactly the same issue with a new set of ESP-01's I was sent from Ali. It seems the new puya flash has some issues. 
I had to apply the same patch to esp.cpp to get the SPIFFS working.
https://github.com/esp8266/Arduino/issues/4061

jmozmoz, 2018-06-04:
@plambrechtsen you are right. Applying the patch https://github.com/esp8266/Arduino/issues/4061#issuecomment-368273656 solves the problem.

timpur, 2018-06-05:
This is interesting to know, sorry I haven't found time to look into this deaply, but when I do, good to know this ... Not sure how to fix this, might be something to add to docks ...

plambrechtsen, 2018-06-05:
I don't think this is something that could easily be fixed in home as it's an underlying issue with the esp8266 SPIFFS writing function. Unless you wanted to lift and shift all of that into Homie which I certainly wouldn't recommend. Otherwise you could add logic into the Web UI and serial console that reads back the config.json file after writing it. If it isn't able to read the config back then you know you have the issue with SPIFFS not working.
Another thought about how this could be worked around is to have an option to write the config into offsets in the EEPROM but that would require agreeing particular offsets for where particular configuration items would sit rather than having a free text json file and being limited to 512/1024 bytes. Which would completely change how config is currently managed today.

jmozmoz, 2018-06-06:
Is it possible to monkey patch that routine (EspClass::flashWrite)?

Otherwise, it would be helpful if it is documented: ESP-01 + PUYA flash => patch your framework.