homieiot/homie-esp8266 #528: Wemos D1 mini, multiple sensors with Homie

sats4eva, 2018-06-12:
Hi Marvin,
I LOVE LOVE LOVE the home framework. I’ve used it to monitor doors with home assistant and it works great. I used an Esp8266-01 black version for this project and it worked well.

I’m tinkering with my wemos d1 mini and i wanted to know if its possible to use multiple sensors on one d1 mini with your framework. To give some context i’m looking to run 2 door sensors and a water sensor all from one wemo d1 mini using different I/O pins. Is this possible wth homie? I was reading through the documentary and I couldn’t find something on multiple sensors or example with the code..

Thanks in advance.

The issue tracker is a great place to ask for enhancements and to report bugs.
If you have some questions or if you need help, some people might help you on the [Gitter room](https://gitter.im/homie-iot/ESP8266).

Before submitting your issue, make sure:

- [ ] You've read the documentation for *your* release (in the `docs/` folder or at http://marvinroger.github.io/homie-esp8266/) which contains some answsers to the most common problems (notably the `Limitations and know issues` and `Troubleshooting` pages)
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples might not be backward-compatible

Thanks!

timpur, 2018-06-12:
You make a good point, we should add that example to the docs.

Yes you can make as many nodes as you need.

`HomieNode temperatureNode("temperature", "temperature");` 

Just keep creating nodes for each sensor and hook them up to the sensor output :)

I normally have many sensor types on one device :)

jmozmoz, 2018-06-12:
Shamelessly pointing to my own project: Here is an example:
https://github.com/jmozmoz/yagds/blob/master/uploadSensorData/src/main.cpp#L30

sats4eva, 2018-06-13:
Awesome thanks. However I’m a bit confused on how to create the Json file. If I have one sensor output how does the software know which tripped it? 
Sorry I’m still somewhat new to this platform but if in the config I call the device “bathroom sensors” and I add a door node and water sensor node, wait as I’m writing it I think I got it lol
Is there a board schematic somewhere?

sats4eva, 2018-06-13:
@jmozmoz what kind of board are you using? ESP-01?

luebbe, 2018-06-13:
@sats4eva , you can also use this project https://github.com/luebbe/homie-node-collection as a starting point. The demo contains plenty of nodes hooked up to different pins. Just pick whatever you like :)

satssehgal, 2018-06-13:
hey guys so i compiled the code but its going into an iterative loop in the serial monitor. can someone spot the mistake? `#include <Homie.h>
#include <ArduinoJson.h>

const int PIN_DOOR = D1;
const int PIN_WATER = D3;

bool sleepFlag = false;

Bounce debouncer = Bounce(); // Bounce is built into Homie, so you can use it without including it first
int lastDoorValue = -1;
int lastWaterValue = -1;

HomieNode doorNode("door", "door");
HomieNode waterNode("water", "water");

void loopHandler() {
  int doorValue = debouncer.read();
  int waterValue = debouncer.read();

  if (doorValue != lastDoorValue) {
     Homie.getLogger() << "Door is now " << (doorValue ? "open" : "closed") << endl;

     doorNode.setProperty("open").send(doorValue ? "true" : "false");
     lastDoorValue = doorValue;
  }
  if (waterValue != lastWaterValue) {
     Homie.getLogger() << "Water sensor state is " << (doorValue ? "Leak Detected" : "clear") << endl;

     waterNode.setProperty("clear").send(doorValue ? "true" : "false");
     lastWaterValue = waterValue;
  }
}

void onHomieEvent(const HomieEvent& event) {
  switch(event.type) {
    case HomieEventType::MQTT_READY:
      Homie.getLogger() << "MQTT connected, preparing for deep sleep..." << endl;
      Homie.prepareToSleep();
      break;
    case HomieEventType::READY_TO_SLEEP:
      Homie.getLogger() << "Ready to sleep" << endl;
      ESP.deepSleep(0);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;
  Homie.disableResetTrigger();  // we do not want to trigger config mode by sensor input
  pinMode(PIN_DOOR, INPUT);
  pinMode(PIN_WATER, INPUT);
  digitalWrite(PIN_DOOR, HIGH);
  digitalWrite(PIN_WATER, HIGH);
  debouncer.attach(PIN_DOOR);
  debouncer.attach(PIN_WATER);
  debouncer.interval(50);

  Homie_setFirmware("my-sensors", "1.0.0");
  Homie.setLoopFunction(loopHandler);
  Homie.onEvent(onHomieEvent);

  doorNode.advertise("open");
  waterNode.advertise("clear");

  Homie.setup();
}

void loop() {

  Homie.loop();
  debouncer.update();
  
}

`

satssehgal, 2018-06-13:
@luebbe I buy bread from my plumber? did you wake up on the wrong side of bed today?
I’m new hear so trying to navigate the rules. Glad to see people here are nice and not jerks :)

luebbe, 2018-06-14:
@satssehgal had a coffee already. Donut would be nice ;)

satssehgal, 2018-06-14:
Air donut on its way :)

ubergeekseven, 2019-06-24:
@luebbe thank you for the code demos. Haven't found very many so far. Still can't find one for neopixel or 8212 LEDs that works with the newer format.