homieiot/homie-esp8266 #529: Getting Infinite loop with compiled code...can you pls help?

satssehgal, 2018-06-13:
Hi Guys,
I’m new here so please dont shoot my head off if I’ve accidentally broken a rule while posting. So i’m new to the homie framework, i’ve successfully compiled code for my door sensor alone but now i want to use additional sensors for my wemos d1 mini. There are obviously a bunch of smart folks on this site and I wanted to seek some assistance. 

I compiled the code but its going into an iterative loop in the serial monitor. can someone please spot where i may have gone wrong? Much appreciated in advance

 `#include <Homie.h>
#include <ArduinoJson.h>

const int PIN_DOOR = D1;
const int PIN_WATER = D3;

bool sleepFlag = false;

Bounce debouncer = Bounce(); // Bounce is built into Homie, so you can use it without including it first
int lastDoorValue = -1;
int lastWaterValue = -1;

HomieNode doorNode("door", "door");
HomieNode waterNode("water", "water");

void loopHandler() {
  int doorValue = debouncer.read();
  int waterValue = debouncer.read();

  if (doorValue != lastDoorValue) {
     Homie.getLogger() << "Door is now " << (doorValue ? "open" : "closed") << endl;

     doorNode.setProperty("open").send(doorValue ? "true" : "false");
     lastDoorValue = doorValue;
  }
  if (waterValue != lastWaterValue) {
     Homie.getLogger() << "Water sensor state is " << (waterValue ? "Leak Detected" : "clear") << endl;

     waterNode.setProperty("clear").send(waterValue ? "true" : "false");
     lastWaterValue = waterValue;
  }
}

void onHomieEvent(const HomieEvent& event) {
  switch(event.type) {
    case HomieEventType::MQTT_READY:
      Homie.getLogger() << "MQTT connected, preparing for deep sleep..." << endl;
      Homie.prepareToSleep();
      break;
    case HomieEventType::READY_TO_SLEEP:
      Homie.getLogger() << "Ready to sleep" << endl;
      ESP.deepSleep(0);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;
  Homie.disableResetTrigger();  // we do not want to trigger config mode by sensor input
  pinMode(PIN_DOOR, INPUT);
  pinMode(PIN_WATER, INPUT);
  digitalWrite(PIN_DOOR, HIGH);
  digitalWrite(PIN_WATER, HIGH);
  debouncer.attach(PIN_DOOR);
  debouncer.attach(PIN_WATER);
  debouncer.interval(50);

  Homie_setFirmware("my-sensors", "1.0.0");
  Homie.setLoopFunction(loopHandler);
  Homie.onEvent(onHomieEvent);

  doorNode.advertise("open");
  waterNode.advertise("clear");

  Homie.setup();
}

void loop() {

  Homie.loop();
  debouncer.update();
  
}

`

this is the error loop
`
Offline message acknowledged. Di✔ Wi-Fi connected, IP: 192.168.1.151
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
MQTT connected, preparing for deep sleep...
Flagged for sleep by sketch
Calling setup function...
Device in preparation to sleep...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 2s
Door is now open
Water sensor state is Leak Detected
Offline message acknowledged. Disconnecting MQTT...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
Triggering READY_TO_SLEEP event...
Ready to sleep

Exception (28):
epc1=0x4022c5c2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000004c depc=0x00000000

ctx: sys 
sp: 3ffffbd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffd70:  00000002 00000000 3ffefea8 40214482  
3ffffd80:  3ffffdb0 00000000 0000033d 40214551  
3ffffd90:  3fff0108 00000001 3ffefea8 402145d0  
3ffffda0:  3fff0108 00000000 3ffefea8 3fff37ac  
3ffffdb0:  3fff0108 3ffef9d8 3ffefea8 40214205  
3ffffdc0:  3fff305c 3ffef9d8 3ffefea8 40212249  
3ffffdd0:  3fff0108 3ffef9d8 3ffefbe4 4020f31a  
3ffffde0:  3fff305c 00000017 3fff304c 40220a3a  
3ffffdf0:  3fff04e4 00000017 3fff304c 402129cb  
3ffffe00:  3fff3772 00000031 3ffefea8 40212a49  
3ffffe10:  3fff3364 3ffffe60 4020de6c 40220a18  
3ffffe20:  3fff3772 00000031 3ffefea8 40220c66  
3ffffe30:  00007fff 00000031 3ffefea8 402138b4  
3ffffe40:  3ffea97c 00000031 3ffefea8 4021291d  
3ffffe50:  3fff3364 0025def1 40211b2c 40220c44  
3ffffe60:  40212a1c 00000000 3ffefea8 00000001  
3ffffe70:  00000021 00004288 00000000 00000002  
3ffffe80:  3fff0029 3fff37ac 00002200 4000050c  
3ffffe90:  3fffc278 40101f8c 3fffc200 00000000  
3ffffea0:  3ffefea8 3fff372c 00000000 40212950  
3ffffeb0:  4000001b 00000030 3fff18b4 40214650  
3ffffec0:  3fff3684 00000001 00000000 3fffd9d0  
3ffffed0:  00000000 00000000 00000000 3fff18d4  
3ffffee0:  3fff3684 3fff18b8 3fff18b4 40214694  
3ffffef0:  3fff3684 3fff18b8 3fff18b9 4022b034  
3fffff00:  dd01a8c0 0000000a 3ffe9d1c 00000000  
3fffff10:  00000000 00000067 3fffff80 3fff374a  
3fffff20:  3fff1a2c 3fff372c 3fff1b14 4022ef59  
3fffff30:  00000014 00000306 00000306 3fff1a2c  
3fffff40:  3fffdc80 3fff1fd4 3fff3364 3fff201c  
3fffff50:  00000008 3fff1a2c 3fff372c 402285b9  
3fffff60:  3fffdc80 3fff1fd4 3fff3364 4010453c  
3fffff70:  40245b5e 3fff1fd4 3fff3364 40245b70  
3fffff80:  3fff373c 3fff372c 00000000 3fff05f0  
3fffff90:  40240a43 00000000 3fff3364 40247acf  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
^H⸮H⸮💡 Firmware my-sensors (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: b4e62d1b5e99
  • Device ID: basement-sensors
  • Name: Basement Sensors
  • Device Stats Interval: 60 sec
  • Wi-Fi: 
    ◦ SSID: mynet
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 192.168.1.200
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? yes
    ◦ Username: not shown
    ◦ Password not shown
  • OTA: 
    ◦ Enabled? yes
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.1.151
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
MQTT connected, preparing for deep sleep...
Flagged for sleep by sketch
Calling setup function...
Device in preparation to sleep...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 2s
Door is now open
Water sensor state is Leak Detected
Offline message acknowledged. Disconnecting MQTT...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
Triggering READY_TO_SLEEP event...
Ready to sleep

Exception (28):
epc1=0x4022c5c2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000004c depc=0x00000000

ctx: sys 
sp: 3ffffbd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffd70:  00000002 00000000 3ffefea8 40214482  
3ffffd80:  3ffffdb0 00000000 00000307 40214551  
3ffffd90:  3fff0108 00000001 3ffefea8 402145d0  
3ffffda0:  3fff0108 00000000 3ffefea8 3fff334c  
3ffffdb0:  3fff0108 3ffef9d8 3ffefea8 40214205  
3ffffdc0:  3fff319c 3ffef9d8 3ffefea8 40212249  
3ffffdd0:  3fff0108 3ffef9d8 3ffefbe4 4020f31a  
3ffffde0:  3fff319c 00000017 3fff318c 40220a3a  
3ffffdf0:  3fff04e4 00000017 3fff318c 402129cb  
3ffffe00:  3fff3772 00000031 3ffefea8 40212a49  
3ffffe10:  3fff306c 3ffffe60 4020de6c 40220a18  
3ffffe20:  3fff3772 00000031 3ffefea8 40220c66  
3ffffe30:  0025b585 00000031 3ffefea8 402138b4  
3ffffe40:  3ffea970 00000031 3ffefea8 4021291d  
3ffffe50:  3fff306c 00000000 40211b2c 40220c44  
3ffffe60:  40212a1c 00000000 3ffefea8 3fffc278  
3ffffe70:  00000021 3fffc200 00000022 3ffea97c  
3ffffe80:  3fff0029 3fff334c 00000000 401006fa  
3ffffe90:  00000030 00000014 ffffffff 00000000  
3ffffea0:  3ffefea8 3fff372c 00000000 40212950  
3ffffeb0:  0000001b 00000000 3fff18b4 40214650  
3ffffec0:  3fff3684 00000001 3fff05f0 00000000  
3ffffed0:  3fffdad0 3fff05e0 00000030 3fff18d4  
3ffffee0:  3fff3684 3fff18b8 3fff18b4 40214694  
3ffffef0:  3fff3684 3fff18b8 3fff18b9 4022b034  
3fffff00:  dd01a8c0 0000000a 3ffe9d1c 00000000  
3fffff10:  00000000 00000067 3fffff80 3fff374a  
3fffff20:  3fff1a2c 3fff372c 3fff1b14 4022ef59  
3fffff30:  00000014 00000343 00000343 3fff1a2c  
3fffff40:  3fffdc80 3fff1fd4 3fff354c 3fff202c  
3fffff50:  00000008 3fff1a2c 3fff372c 402285b9  
3fffff60:  3fffdc80 3fff1fd4 3fff354c 4010453c  
3fffff70:  40245b5e 3fff1fd4 3fff354c 40245b70  
3fffff80:  3fff373c 3fff372c 00000000 3fff05f0  
3fffff90:  40240a43 00000000 3fff354c 40247acf  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<
`

timpur, 2018-06-13:
I haven't tested the code yet, but by looking at it, it looks like your doing to much to soon and not enough time given to the IP stack to actually send all the packets, not sure about this though.

Will test the code later.

satssehgal, 2018-06-14:
@timpur thanks Tim. So maybe this can help zero in on the issue. when i comment out Homie.onEvent(onHomieEvent); it compiles and works fine, it just doesn't enter deep sleep.
Also i changed ESP.deepsleep() to Homie.doDeepSleep() and the loop went away but now it just crashes after a few min. I just dont know what im doing wrong there.

Sorry what do you mean decide stack trace?

luebbe, 2018-06-14:
@satssehgal Thanks for opening a new ticket. You still need some practice how to format source code in a ticket though. ;) Using preview and the github markdown cheat-sheet might get you up and running.

I think @timpur wanted to write "decode" stack trace. For instance with one of these tools: https://github.com/me-no-dev/EspExceptionDecoder
https://github.com/littleyoda/EspStackTraceDecoder

Does you code run, when you leave out the sleep stuff?

luebbe, 2018-06-14:
Can you really attach two pins to one debouncer? I haven't checked, but I don't think so.
What do you think will happen when your ESP is sleeping and the door opens/closes or your water pin changes state?

satssehgal, 2018-06-14:
You’re totally right
Sorry I should have stated that I created another denounce. I’ll post the updated code tonight when I’m home. That was the other change I made

timpur, 2018-06-14:
Found updating the `AsyncMqttClient` lib fixed this for me, `pio lib install AsyncMqttClient`

n8twj, 2018-06-14:
Also, make sure GPIO16 is connected to RESET, to be woken up from deep
sleep.

satssehgal, 2018-06-14:
Thanks I actually didn’t do that. Would you suggest I solder a wire between DO and RST on the D1 mini or use a resistor? 

For the ESP-01 can someone correct me if I’m wrong in saying that gpio is that really tiny pin just above the ground pin? Has anybody been able to solder that?

satssehgal, 2018-06-14:
Will try that too. Thanks :)

satssehgal, 2018-06-14:
I have version 0.8.2 installed so it looks like it’s updated. I still have the issue but will try the reset to D0 button connected

satssehgal, 2018-06-14:
@n8twj @timpur so tying D0 and RST didnt fix the situation. In node red I have the correct status reporting but it continues to report every second. In serial monitor Arduino IDE, same crazy loop.

Any other suggestiosn?

timpur, 2018-06-14:
@satssehgal 
Try the latest everything? 2.4.1 Arduino core also ?

satssehgal, 2018-06-14:
@timpur yeah I’m running 2.4.1
Still same issue 

My board is wires as follows. Maybe it’s my config

I’ll just put the config for the door sensors, water sensor is the same except different pin

D0 —-> RST
Door sensor lead 1 —-> 3V3
Door sensor lead 2 ——> D1
D1 —-> 4.7k resistor —-> Ground ( I’ve tried this with a straight wire and resistors from 2.2k to 10k)

I’m sure it’s something silly I’ve missed

satssehgal, 2018-06-14:
@timpur the code compiles fine but when I put it on the D1 mini it goes through that crazy loop. :(

satssehgal, 2018-06-15:
BTW here are my stacktrack results...again i have the latest libs and arduino core....

```ruby

Decoding stack results
0x402145a6: std::function ::operator()(void*, AsyncClient*) const at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2465
0x40214675: AsyncClient::_close() at /Users/MBP/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp line 339
0x402146f4: AsyncClient::close(bool) at /Users/MBP/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp line 206
0x40214329: AsyncClient::send() at /Users/MBP/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp line 266
0x4021236d: AsyncMqttClient::disconnect(bool) at /Users/MBP/Documents/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp line 709
0x4020f44a: HomieInternals::BootNormal::_onMqttPublish(unsigned short) at /Users/MBP/Documents/Arduino/libraries/homie-esp8266-2.0.0/src/Homie/Boot/BootNormal.cpp line 555
0x40220b62: std::_Function_handler    (HomieInternals::BootNormal*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, unsigned short) at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x40212aef: std::function ::operator()(unsigned short) const at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2465
0x40212b6d: AsyncMqttClient::_onPubAck(unsigned short) at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2174
0x4020df9c: std::_Function_base::_Base_manager    (HomieInternals::BootNormal*, std::_Placeholder1>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1931
0x40220b40: std::_Function_handler    (HomieInternals::BootNormal*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, unsigned short) at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
0x40220d8e: std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, unsigned short) at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x402139d8: AsyncMqttClientInternals::PubAckPacket::parseVariableHeader(char*, unsigned int, unsigned int*) at /Users/MBP/Documents/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient/Packets/PubAckPacket.cpp line 25
0x40212a41: AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int) at /Users/MBP/Documents/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp line 436
0x40211c50: std::_Function_base::_Base_manager    (AsyncMqttClient*, std::_Placeholder1>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1931
0x40220d6c: std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, unsigned short) at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
0x40212b40: AsyncMqttClient::_onPubAck(unsigned short) at /Users/MBP/Documents/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp line 576
0x40212a74: std::_Function_handler ::_M_invoke(const std::_Any_data &, void *, AsyncClient *, void *, unsigned int) at /Users/MBP/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x40214774: AsyncClient::_recv(tcp_pcb*, pbuf*, long) at /Users/MBP/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp line 415
0x402147b8: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at /Users/MBP/Documents/Arduino/libraries/ESPAsyncTCP-master/src/ESPAsyncTCP.cpp line 493
0x4022b15c: tcp_input at core/tcp_in.c line 435
0x4022f081: ip4_input at core/ipv4/ip4.c line 685
0x402286e1: ethernet_input_LWIP2 at netif/ethernet.c line 182
0x4010453c: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 433
0x40245c86: ethernet_input at glue-esp/lwip-esp.c line 357
0x40245c98: ethernet_input at glue-esp/lwip-esp.c line 365
```

timpur, 2018-06-15:
That's the error I got, but updated my mqtt lib and it went away ...

satssehgal, 2018-06-15:
@timpur yeah it’s really odd I can’t seem to get it to work. My mqtt lib is updated. For me I’m not sure that is the issue because if I remove the deepsleep function it works without issue

luebbe, 2018-06-15:
There are quite a lot of matches when you google for "[Exception  (28)](https://www.google.com/search?q=Exception+(28))"
https://arduino-esp8266.readthedocs.io/en/latest/exception_causes.html

@satssehgal did you delete all previous version of your libs inside the library path? I sometimes had trouble when an old version was found alongside a new library. In PlatformIO I delete the entire lib folder every now and then and let it rebuild, when I get spurious errors.

satssehgal, 2018-06-15:
@luebbe thanks for the suggestion. I just did that and still didnt work.

I had exception 28 - A load referenced a page mapped with an attribute that does not permit loads.
Dont even know what means :)

luebbe, 2018-06-15:
I think it means that a pointer is pointing to nirvana an something is trying to read/write from that location.

satssehgal, 2018-06-15:
@luebbe i'll have to research the error 28...i'm just baffled how some of you got it to work while mine is throwing off this error still.
I just tried to compile it with a new D1 mini and same issue.

timpur, 2018-06-15:
@satssehgal , can you give us a break down of your environment (Platform IO?, arduino core version? ...ect), board your use and flash size ... everything ?

Have you set the right flash size, the right flash mode ect?

satssehgal, 2018-06-15:
@timpur

Here are the details: 

Board : Wemo D1 mini
Env: Arduino Core 2.41
Tool: Arduino IDE
Flash size: 4M (1M SPIFFS)
Debug Port: Disabled
Debug Level: None
IwIP Variant: v2 Lower Memory
CPU Frequency: 80 MHz
Upload Speed: 115200
Erase Flash: Only Sketch
Port: Port on Comp
Programmer: AVRISP mkill

Anything else?

timpur, 2018-06-17:
@satssehgal , thanks. Ive clean my libs in arduino and started from scratch with latest libs of everything needed for homie and have, your issue again so ill solve this once and for all :) might take me a bit though ....

satssehgal, 2018-06-17:
@timpur
No problem 
I appreciate your help. Thanks :)

satssehgal, 2018-06-22:
Anybody make any headway?
So it looks like it breaks even if i comment out deepsleep. Soon as it breaks from the case statement it breaks no matter what is in the case statement

timpur, 2018-06-23:
Thanks for sharing that, im going to look at it over the weekend, sorry bussy week ...

satssehgal, 2018-06-23:
No problem. I’ve been racking my brain with this one for a while. Thanks for your help

satssehgal, 2018-07-03:
> I also got this problem. I worked around it, by disabling all mqtt publishing and wait for a certain amount of time before issuing the command to go to deep sleep. Have a look at the source code:
> https://github.com/jmozmoz/yagds/blob/master/uploadSensorData/src/main.cpp
>
> Actually I do not call Homie.loop() anymore during waiting to go to deep sleep.
>
> I experimented a lot and this is my current solution which seems to work.

Thanks and how is battery life for you on this?

gaelgrasset, 2018-07-03:
I had the exact same issue last week... Oddly it was  out when compiled with Arduino IDE. Debug stack pointed out that Async MQTT was the source of the problem, but I was on the last version of all libs. I found 2 different workarounds:
- revert Homie on Arduino IDE to 2.0.0 beta 2 version (which appears to execute as intended... But you have to update the code, ie Homie.doDeepSleep() is not available)
- OR change IDE to PlatformIO and use latest stable  Homie in lib_deps  (it is now my main IDE, way better than Arduino IDE!). PIO manages automatically the dependances and seems to compile a bit different than Arduino IDE.

jmozmoz, 2018-07-03:
I am using PlatformIO and see this problem. Here my platformio.ini:
```
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
build_flags = -DASYNC_TCP_SSL_ENABLED=1 -Wl,-Teagle.flash.1m64.ld
lib_deps =
  https://github.com/marvinroger/homie-esp8266.git#v2.1
  simpleTimer
  jled
```

satssehgal, 2018-07-26:
@gaelgrasset I tried both methods and neither seem to work. I’ve just been using an outlet over batteries out of sheer frustration

x821938, 2018-10-03:
I have also been fighting with this problem.
My solution was to revert to 0.8.1 of  AsyncMqtt: https://github.com/marvinroger/async-mqtt-client/releases/tag/v0.8.1 (remember to build clean and rebuild everything)

MoJo2600, 2019-04-21:
I ran in the exact same problem. Reverting AsyncMqtt to 0.8.1 resolved the error for me.

boneskull, 2019-07-04:
FWIW I couldn't figure out how to revert to v0.8.1 using PlatformIO without vendoring Homie and manually patching its `library.json`.  I didn't love that idea, but thankfully it wasn't necessary.

Instead, I used @jmozmoz's strategy [above](https://github.com/homieiot/homie-esp8266/issues/529#issuecomment-401976023)--once I was able to get my head around it--and get consistent deep-sleep *without exceptions* on a D1 Mini Pro. The trick, as @jmozmoz mentions, is to essentially *stop* the Homie loop once `Homie.prepareSleep()` has been called.  But once you stop the loop, the `READY_TO_SLEEP` event either won't emit or the handler won't be invoked (not sure which), so you need to manually invoke deep sleep via a timer (`sleepTimeoutId` in @jmozmoz's code).  

I'm not sure how short that sleep timeout can be, but if it's too short, I imagine Homie wouldn't be fully "ready" for sleep when `Homie.deepSleep()` is called.  I'm also unsure of the necessity of the subsequent `ESP.deepSleep()` call.

I combined this with the wifi reset strategy [mentioned here](https://github.com/esp8266/Arduino/issues/3072#issuecomment-348692479), because wifi was not reconnecting after wake from deep sleep.  To mitigate, it looks like you just need to call `WiFiOff()`, delay a short period, then call `WiFiOn()` again at the start of your `setup()` function.

MoJo2600, 2019-07-20:
Just for the first part on reverting to v0.8.1. The order of the libs matters in the platformio.ini. This is how mine looks like:

```
lib_deps = 
    AsyncMqttClient@0.8.1,!=0.8.2
    ArduinoJson@5.13.4
    https://github.com/homieiot/homie-esp8266.git#develop-v3
    Timer
```