homieiot/homie-esp8266 #531: Add link to OTA config update to the config section

luebbe, 2018-06-25:
When trying to find information about the configuration of a homie device one naturally looks in the "configuration" section of the docs. The info about updating config.json via OTA/MQTT however is found in the "others" section.
A link from "Configuration" to the section under "Others" would be helpful.
This is where I looked: http://marvinroger.github.io/homie-esp8266/docs/2.0.0/configuration/json-configuration-file/
And this is where the info actually is: http://marvinroger.github.io/homie-esp8266/docs/2.0.0/others/ota-configuration-updates/

igrowing, 2018-10-08:
Hello guys. I apologize posting here a question. I've read the documentation about OTA, links above. 
My setup is:
Workstation -- MQTT-broker -- Homie-over-ESP.
10.100.101.100 -- 10.100.101.5 -- 10.100.101.50.

From the Workstation I tried to use following commands + results:
`python ota_updater.py -l 10.100.101.10 -i rls0 .pioenvs\esp12e\firmware.bin
Connecting to mqtt broker 10.100.101.5 on port 1883
Connected with result code 0
Waiting for device to come online...`
No more progress. I tried to reboot the ESP. Broker sees the ESP as:
`shm/rls0/$stats/interval 0
shm/rls0/$fw/name awesome-relay
shm/rls0/$fw/version 1.0.0
shm/rls0/$fw/checksum 1d586180790690710f5c0d65f8198024
shm/rls0/$implementation esp8266
shm/rls0/$implementation/config {"wifi":{"ssid":"trololo"},"mqtt":{"host":"10.100.101.10","port":1883,"base_topic":"shm/","auth":false},"name":"remote light 0","ota":{"enabled":true},"device_id":"rls0"}
shm/rls0/$implementation/version 2.0.0
shm/rls0/$implementation/ota/enabled true
shm/rls0/light/$type switch
shm/rls0/light/$properties on:settable
shm/rls0/$online true
shm/rls0/$stats/signal 78
`

An attempt to use direct IP of ESP fails too:
`python ota_updater.py -l 10.100.101.50 -i rls0 .pioenvs\esp12e\firmware.bin
Connecting to mqtt broker 10.100.101.50 on port 1883
Traceback (most recent call last):
  File "ota_updater.py", line 172, in <module>
    args.broker_password, args.broker_tls_cacert, args.base_topic, args.device_id, firmware)
  File "ota_updater.py", line 119, in main
    client.connect(broker_host, broker_port, 60)
...bla-bla-bla...
socket.error: [Errno 10061] No connection could be made because the target machine actively refused it`

Port 1883 in Homie is not listening, however Homie is reporting the status every minute to the broker.
I tried to perform regular OTA from platformio (standard Arduino procedure), fails too since port 8266 isn't listening too.

I think maybe my setup is wrong. What it should be?
Or maybe this is because I use non-homie base topic?
I use latest Homie 2.0.0 from branch development (last check in 4 months ago).
Please, help.

jmozmoz, 2018-10-08:
I think you have to use the IP address of the broker to publish the OTA update: 10.100.101.5 
(not 10.100.101.50). 

Which ESP8266 do you use? Make sure, that there is enough space available to (temporarily) store the OTA. 

Have look at the settings for -Wl,-Teagle.flash.xxx
http://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size

igrowing, 2018-10-09:
Thank you, Joachim for confirming my understanding of the OTA process is correct.
I tried Homie OTA on 2 modules:
ESP-12e compiled with flags: `-Wl,-Teagle.flash.4m.ld`
X-8266 (based on Wroom-02 with 2MB Flash) compiled with flags: `-Wl,-Teagle.flash.1m512.ld`

For both modules it is more than enough room in SPIFFS for OTA.
If the space would a problem, I'd expect according error "Not enough space" or similar. In given case there is not any reaction.

stritti, 2019-04-09:
documentation extended by suggested reference (branch `develop-v3`).

closed 2019-04-09 by stritti