homieiot/homie-esp8266 #534: Logging to MQTT

mosseb, 2018-07-22:
Hi.

I'm nearly reading to put my homie powered module in the attic.
When it's there, I want it to be monitored from my desktop pc, obviously.
Updating firmware with OTA MQTT is perfectly fine and works OK.
Homie is outputting a lot of usefull logs on the serial, but when it's up there, I will not have access to it anymore.
If there a possibility to redirect logging to MQTT to have a full access to it, wherever the module is ?
According to doc, I don't think this is not something that exists, or plan to exist today.

We could have a topic homie/deviceid/$log/active, when true it would log on serial + mqtt
Then a second topic homie/deviceid/$log/console where we output each line one by one.

I'm not really sure if it is easily feasible, but I suppose it would not be that hard ?
Are you interested in such a possibility ?

Thanks for your feedback.

euphi, 2018-07-23:
FYI: I did some logging mechanism here https://github.com/euphi/HomieLoggerNode .

However, this does not log Homie internals (yet).

mosseb, 2018-07-24:
I think this could be something relatively easy and usefull to add to Homie itself.
Any plan for that @marvinroger, @timpur, @ThomDietrich ?

ThomDietrich, 2018-07-24:
https://github.com/homieiot/convention/issues/77#issuecomment-385789150

> Why do we need log topics? Why do we need them as fixed part of the convention? If correctly configured, your device offers the right set of nodes and properties to publish all relevant data that you'd otherwise dump to a log. A well implemented node doesn't need a log.

> For debugging purposes a regular node/property can be used.

mosseb, 2018-07-24:
I really don't agree that we "don't need log if a node is well implemented".
By definition, we have bug in software.
Logs are a good way to obtain info about what happens.
If you say we don't need log over mqtt (which is the exclusive way of communicating with the device), then explain me why we need log when in serial :) This is non sense to me.
I think discussion should not be whether we need log or not, but what is the better way to access those logs with MQTT, absolutely including homie internal logs.
Nevertheless, thanks for pointing me to homieiot/convention#77 where I can see this is already in great debate.
May I close this topic or is this still a pending subject for marvinroger/homie-esp8266 ?
Thank you.

ThomDietrich, 2018-07-24:
Providing an infrastructure for logs in the homie convention does not make sense, because it is supposed to support the communication of information and commands and the means for auto-discovery of those by a controller.

That said, I agree that a mean for debugging is useful during development and testing. As I said, you can simply define a node/property for the purpose and publish as much verbosity as you like there. I have a solution just like it: https://github.com/ThomDietrich/SenseoWifi/blob/master/src/SenseoWifi.cpp#L172

Yes, feel free to close this issue and join https://github.com/homieiot/convention/issues/77 instead 😉

mosseb, 2018-07-24:
I don't quite understand why this is not related to homie convention.
I don't feel like my custom logs are sufficient.
I'd wish all homie logging are pushed to the same channel.
I don't understand why one would not like to have mqtt disconnections, resets, wifi statistics.. as they are anyway produced to be put on the serial port.
Concerning my custom logs, I agree about the kind of implementation you're proposing.
Thank you.

ThomDietrich, 2018-07-25:
Because the convention concerns itself with application oriented data and actions and the auto-discovery of these. 😃 
It does not disallow or discourage the implementation of a debugging channel. It's simply out of the main scope. That said, now that you are also speaking about disconnects or wifi stats, please see https://github.com/homieiot/convention/issues/102 (and probably others. Join)

ThomDietrich, 2018-07-25:
Addition: What I said is related to the convention. I'm not sure if the "homie-esp8266" framework should offer some options related to statistics. @timpur

mosseb, 2018-07-25:
I would like to have @marvinroger opinion concerning homie-esp8266.
Actually, it might be a little bit more difficult than just push a MQTT message as long as a serial message : if wifi or mqtt is down, how would you log it ? For it to work as expected, it has to memorize logs when mqtt is not reachable. Not sure this is actually relevant after all. I'm not MQTT specialist but what I understand is the messages are lost unless something else scrutates for it (not true only for the last one if retain flag is set).

marvinroger, 2018-07-25:
@mosseb what you're asking is already doable thanks to [events](http://marvinroger.github.io/homie-esp8266/docs/stable/advanced-usage/events/). You can then use, as said above, a logging node.

mosseb, 2018-07-25:
OK so everything is here to do it, got it.
Thanks for the advice and all comments of every people.

closed 2018-07-25 by mosseb

timpur, 2018-09-09:
@mosseb , not sure if this interests you https://github.com/euphi/HomieLoggerNode

mosseb, 2018-09-09:
Yes this is interesting.
Concerning issues when MQTT or Wifi is erratic, I dont' think there is any intelligent thing to do at module side. We should log this by historizing the alive flag, in a home automation system software for example.
This answers my question fully, thanks for sharing :)