homieiot/homie-esp8266 #535: Update error: ERROR[4]: Not Enough Space

WaJoWi, 2018-07-29:
I get this error after uploading (OTA) ESPEasyUploaderMega_1024.bin.

Tested with GIT version: mega-20180710 and mega-20180722 (two different ESP8266).

Is the uploader image too large?

timpur, 2018-07-29:
What's the size of the bin ? 1MB ? 

You sketch can only be as big as half the size of your device storage, because half is for current sketch and the other half of the flash is used as temp storage for ota. (Not entirely accurate ***)

Thus if your device storage is 1MB then your sketched need to be less than half to ensure there is room for the next ota sketch ... 

This might be the case with your situation

WaJoWi, 2018-07-29:
I mentioned that I wanted to flash the uploader-bin. This one has 270k! Currently I have the 1024k-bin running.

Flash chip real size: 4096 kB
Flash IDE size: 1024 kB

euphi, 2019-01-22:
How is this related to homie-esp8266?

If it is and still an issue, please re-open.

closed 2019-01-22 by euphi