homieiot/homie-esp8266 #536: Homie ESP-8266 is not working without wifi

kmehran, 2018-09-14:
The issue tracker is a great place to ask for enhancements and to report bugs.
If you have some questions or if you need help, some people might help you on the [Gitter room](https://gitter.im/homie-iot/ESP8266).

Before submitting your issue, make sure:

- [ ] You've read the documentation for *your* release (in the `docs/` folder or at http://marvinroger.github.io/homie-esp8266/) which contains some answsers to the most common problems (notably the `Limitations and know issues` and `Troubleshooting` pages)
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples might not be backward-compatible

Thanks!

kmehran, 2018-09-14:
I designed a wifi light-switch, based on Homie and Openhab Server, the first important function of 
each light switch is turning on and off the lights, but it cannot turn on/off the light during wifi/mqtt
connection or even when one or both of them lost.
One of the Most important matter about adding extra functionality or new technology to an old 
system (like Light Switch) is do not affect its basic function(s).

Alex9779, 2018-10-22:
I found this somewhere else here, maybe in another issue description, the solution is depending on your implementation that you run the loop where you handle the events the manual switch raises in the mail loop in addition to the Homie.loop().
Here an excerpt of one of my implementations:
```
void loop()
{  
  Homie.loop();

  button.tick();
}
```
In this firmware is use the lib "OneButton" for handling a button, it is a uplight control firmware and the uplight has one button where you can turn it on and off in addtion to control it with Homie.

kmehran, 2018-10-22:
> You can see the "loop" for handling the button events is also in the main loop where the Homie.loop() is too. So both are run and no matter how the Homie state is the button events are processed...

Dear Alexander,

That should be my solution, thanks for your
complete answer.

Regards,
Kaveh.

euphi, 2019-01-22:
See here, how to change the Homie-ESP8266 sources to get your desired behavior:

https://github.com/homieiot/homie-esp8266/pull/490/files

https://github.com/homieiot/homie-esp8266/blob/4e1cfe583c1e067928f874ee7ef7e04b3dd4c8d0/src/Homie/Boot/BootNormal.cpp

Misiu, 2019-02-05:
@euphi, as this is labeled with `Feature_V3`, does it means that this will be added to develop-v3 branch?
I'm trying to build Wemos D1 based multi-relay switch and recently I found this repo. Because OpenHab support homie 3.x I want to use this library.
As many wrote before sometimes we don't want to wait for MQTT to connect at the start, the most important thing is the working relay.

I've found an example that I want to use as a base for my sketch - https://github.com/homieiot/homie-esp8266/blob/develop-v3/examples/IteadSonoffButton/IteadSonoffButton.ino

How should I change it so that I can toggle replay when I don't have MQTT connected?
Example showing such behavior would be awesome.

euphi, 2019-03-10:
Closed with #490.

closed 2019-03-10 by euphi