homieiot/homie-esp8266 #537: What version of ESPAsyncTcp to use?

duke1swd, 2018-09-16:
Getting back to working on a homie-based project after some time away, and using a newly imaged laptop.  Installed all dependency libraries as per the getting started documentation for version 2.0.0.  Got a number of compile errors.  Upgraded my ESPAsyncTcp from the specified commit (c8ed544) to the most current commit (7e9ed22) and now it compiles.  Have yet to try running anything.

Can we get a tagged version of ESPAsyncTcp that we should use?

ccarrizosa, 2018-11-04:
Hi,

I got the same problem and setting ESPAsyncTcp to v1.2.0 is working for me.

stritti, 2019-03-30:
ESPAsyncTCP v1.2.0 is currently used as dependency of AsyncMqttClient v0.8.2.

closed 2019-03-30 by stritti