homieiot/homie-esp8266 #538: Crash when going into deep sleep with NodeMCU

Zero545, 2018-09-20:
Hello,

I am using homie-esp8266 release 2.0.0. I just tried to use deepsleep on a NodeMCU I just got. Unfortunatly I am not able to get it working together with Homie. Without Homie deepsleep is working.

This is the code I used:

```
#include <Homie.h>

void onHomieEvent(const HomieEvent& event) {
  switch(event.type) {
    case HomieEventType::MQTT_READY:
      Homie.getLogger() << "MQTT connected, preparing for deep sleep..." << endl;
      Homie.prepareToSleep();
      break;
    case HomieEventType::READY_TO_SLEEP:
      Homie.getLogger() << "Ready to sleep" << endl;
      Homie.doDeepSleep();
      break;
  }
}

void setup() {
  pinMode(D4, OUTPUT); // Port aus Ausgang schalten
  Homie.setLedPin(D4, LOW);
  Serial.begin(115200);
  Serial << endl << endl;
  Homie.onEvent(onHomieEvent);
  Homie_setFirmware("nodemcu-sound-sensor", "1.0.0");
  Homie.setup();
}

void loop() {
  Homie.loop();
}

```
And these are the outputs I get:

```
💡 Firmware nodemcu-sound-sensor (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: bcddc2147dd4
  • Device ID: nodemcu-sound-sensor
  • Name: NodeMCU Sound Sensor
  • Device Stats Interval: 60 sec
  • Wi-Fi: 
    ◦ SSID: edesteil
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 192.168.0.39
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? yes
    ◦ Username: eweis
    ◦ Password not shown
  • OTA: 
    ◦ Enabled? yes
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.0.45
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
MQTT connected, preparing for deep sleep...
Flagged for sleep by sketch
Calling setup function...
Device in preparation to sleep...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 0s
Node sound not registered
Offline message acknowledged. Disconnecting MQTT...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
Triggering READY_TO_SLEEP event...
Ready to sleep
💤 Device is deep sleeping...

Exception (28):
epc1=0x4022d2d6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000004c depc=0x00000000

ctx: sys 
sp: 3fffeb40 end: 3fffffb0 offset: 01a0
```

Does anyone know what could be going wrong in this easy example?

x821938, 2018-10-03:
I am also scratching my head with this on. I have exact the same problem.

x821938, 2018-10-03:
I have also been fighting with this problem.
My solution was to revert to 0.8.1 of AsyncMqtt: https://github.com/marvinroger/async-mqtt-client/releases/tag/v0.8.1 (remember to build clean and rebuild everything)

Zero545, 2018-10-06:
> 
> 
> I have also been fighting with this problem.
> My solution was to revert to 0.8.1 of AsyncMqtt: https://github.com/marvinroger/async-mqtt-client/releases/tag/v0.8.1 (remember to build clean and rebuild everything)

Thank you for the hint. I also had success with version 0.8.1 👍

closed 2018-10-06 by Zero545

nuccleon, 2018-10-18:
I have the same problem ^^
And this is my workaround:

```javascript
void onHomieEvent(const HomieEvent& event) {
  static bool readyToSleep = false;
  switch (event.type) {
    case HomieEventType::READY_TO_SLEEP:
      WiFi.disconnect();    // Calling doDeepSleep() at this point sometimes causes exceptions due to asynchronous arrival 
                            // of WIFI_DISCONNECTED. WDT wakes up the Device immediatly after the exception. So deep sleep does not work.
                            // To work around this, I decided to inherit this task to the WIFI_DISCONNECTED event.
      readyToSleep = true;
      break;
    case HomieEventType::WIFI_DISCONNECTED:
      if(readyToSleep)
        Homie.doDeepSleep(publishIntervalSetting.get() * 1000000);
      break;
  }
}
```

Alex9779, 2018-12-16:
I ran into the same problem a day ago and tried everything though I wasn't able to get VSCode and PIO use the old lib of AsyncMqttClient I search in that code what change and reverting https://github.com/marvinroger/async-mqtt-client/commit/b4d2c88cb5dc4ebaebcc61b22df014993d1d105a makes the whole thing working for me again...
As another approach I moved the line introduced in that commit up by two lines, no realy idea of whats going on but just by reading the code, sending after disconnecting doesn't seem to be a good idea...
And that way it works too...

markuspalme, 2019-07-10:
This problem still exists.