homieiot/homie-esp8266 #541: Sonoff Basic configuration issue

3434dfgdggh7, 2018-10-07:
Hello, I have a problem that keeps me busy for some days. I have several Homie devices running @version 2.0. Thank you for creating Homie!

I try to setup a Sonoff Basic with Homie, flashing works fine, but I can’t get the device configured with the config.json file. 

If I configure the device and it reboots into normal mode I get these errors in exactly these order:

I write the config.json with curl after the first reboot that follows upon a fresh flash (every write described below succeeds with {"success":true} ):

✔ Configured
↻ Rebooting into normal mode...
✖ /homie/config.json doesn't exist

Second try, writing config.json with curl again:

✔ Configured
↻ Rebooting into normal mode…
✖ Config file too big

Third try, writing config.json with curl again:

✔ Configured
↻ Rebooting into normal mode…
✖ Invalid JSON in the config file

Fourth try, writing config.json with curl again:

✔ Configured
↻ Rebooting into normal mode…
✖ Invalid JSON in the config file

Starting from the third write only the message “✖ Invalid JSON in the config file” shows up. This is 100% reproducible for both the Arduino IDE and PlatformIO.

- The same behavior is seen when uploading config.json directly into flash with the SPIFFS data upload tool (same behavior for ui_bundle, its not detected upon start).
- I’m 100% sure that the config.json file is OK, I use it for other devices and it is correctly parsed when uploaded. 
- I tried every flash size setting with every SPIFFS size (1M with 64K SPIFFS should be ok).
- I tried every flash mode setting: QIO, QOUT, DIO, DOUT (DOUT should be ok).
- I’m using the generic ESP module but also tried NodeMCU 0.9 etc.
- I used different ESP core Arduino versions.
- I used several USB to serial convertors (FT232R board from Sparkfun works best).
- I completely erase the flash before flashing.
- I’m using the minimal Homie sketch, to exclude any other potential issue. 

Minimal Homie: 

#include <Homie.h>

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;

  Homie_setFirmware("bare-minimum", "1.0.0"); 

  Homie.setLedPin(13, LOW);
  Homie.setup();
}

void loop() {
  Homie.loop();
}

Config.json:
{
  "name": "i034-son-basic",
  "device_id": "i034",
  "device_stats_interval": 60,
  "wifi": {
    "ssid": "XXXXXX",
    "password": "XXXXXXX"
  },
  "mqtt": {
    "host": "192.168.128.7",
    "port": 8883,
    "base_topic": "athome/",
    "auth": true,
    "username": "XXXXXXX",
    "password": "XXXXXXX"
  },
  "ota": {
    "enabled": true
  }
}

timpur, 2018-10-08:
Can you try uploading the config.json via spiffs image flashing

timpur, 2018-10-08:
This should help: http://marvinroger.github.io/homie-esp8266/docs/2.0.0/configuration/json-configuration-file/

carlosarmentac, 2018-11-30:
I have the same issue, when I upload tue config via spiffs works ok. But if I delete the config trough Reset Button, then the sonoff enter in config mode, and the issue ocurs again.