homieiot/homie-esp8266 #544: To Everyone: My future support

timpur, 2018-10-22:
Sorry guys.

Over the last year I've found my self having berry little time due to finishing uni and starting a full time job. With that lack of time, I've been questioning how I can apply my time in the most efficient way.

My thoughs are to move to a higher level language. ATM I'm looking at micropython.

I will commit what I have locally and slowly transition.

I'm still committed to Homie, just in a different way. 

Sorry guys.

amayii0, 2018-10-23:
Hi @timpur 
Life's made of priorities and that's normal.

However I just found this while googling your report:
https://github.com/microhomie/microhomie

Regards

timpur, 2018-10-24:
That's the one I'll probs bee working on next ....

bodiroga, 2018-10-25:
Hi @timpur!

Oh, I'm sorry to hear that, but good luck with your full time job! 😉

Let's hope that this won't be the end of homie-esp8266!

Best regards,

Aitor

bertmelis, 2019-04-04:
In the near future (starting July) I'm on leave of absence so I have time to support this project...