homieiot/homie-esp8266 #545: SonOff blinks like MQTT disconnect - but is controlable over MQTT

JorinL, 2018-10-22:
I encounter the following problem with ALL my SonOffs (1x Dual, 8x S20, 2xBasic):

They all start blinking like they lost their MQTT connection. But i can switch em via MQTT.

After a reboot they are all working like nothing did ever happen.

I currently use esp 8266 version 2.4.2 - could it be, that something new inside there is currently not handled with homie 2.0.0? Or is it a Bug in Homie?

To define where this comes from I installed now a test set:

1x S20 with HOMIE 2.0.0 and ESP 2.4.1 (had the blinking thing with 2.4.2)
1x S20 with HOMIE 2.0.0 and ESP 2.4.2 (unchanged - will start blinking)
1x Touch with HOMIE 2.0.0 and ESP 2.4.1 (had the blinking thing with 2.4.2)

But the handling of the LED is pure Homie as far as i know. 
I also have some S20 in a kind of "nightmode" where the LED feedback is disabled - so there is nothing with that problem. But, not on every device I wan't no feedback.

henrik-muehe, 2018-12-13:
I unfortunately observe the exact same issue.

awilliam, 2018-12-13:
Me too!  Using the generic esp8266 module target for a TYWE3S found in a KMC smart switch, after only a few hours the LED would start blinking indicating disconnected from the mqtt server.  Homie on the device was still checking in uptime, but the device would stop sending property updates.  Therefore using a node-red switch, I could, for example, turn the device off, but the off status is not published by the device so the switch in the dashboard doesn't reflect the current state and I can no longer control the device via the ui since the switch is in the opposite state from the device.  Interestingly, restarting the mosquitto mqtt server would allow the device to reconnect, but no matter what I did the device would start flashing again before long.  Using the board manager to downgrade to esp 2.4.1 seems to resolve the issue.

JorinL, 2018-12-14:
Ah thought I'm alone.

I can say it is not fixed by using 2.4.1 or 2.4.2.
It seems that the main problem is, that it looses wifi while mqtt connection ist los - but first the mqtt is lost then the wifi. So that it trys to reconnect to mqtt and not to wifi (seems to be a blocking thing)

But I use 2.4.2 and fixed my problem to 100% using the following lines of code:

`void onHomieEvent(const HomieEvent & event) {
  switch (event.type) {
    case HomieEventType::WIFI_CONNECTED:
      // Do whatever you want when Wi-Fi is connected in normal mode
      ArduinoOTA.setHostname(Homie.getConfiguration().deviceId);
      ArduinoOTA.begin();
      // You can use event.ip, event.gateway, event.mask
      break;
    case HomieEventType::WIFI_DISCONNECTED:
      // Do whatever you want when Wi-Fi is disconnected in normal mode
      WiFi.disconnect();
      WiFi.begin();
      // You can use event.wifiReason
      break;
    case HomieEventType::MQTT_DISCONNECTED:
      // Do whatever you want when MQTT is disconnected in normal mode
      WiFi.disconnect();
      WiFi.begin();
      // You can use event.mqttReason
      break;
  }
}`

And inside the setup:

`Homie.onEvent(onHomieEvent); // before Homie.setup()`

JorinL, 2018-12-27:
Mh.. I have to set those 100% to 85% :-)

Sometimes it still happens...

kleini, 2019-01-01:
First of all, Homie is a great library to ease software development for ESP 8266 devices.

I have the same problem with two Sonoff Basic devices. I have to admit, that I use the V2.1 version in development from https://github.com/marvinroger/homie-esp8266/pull/458. I like to secure my MQTT connections using TLS.

The TLS connection to MQTT does not come properly up in the first attempt and this often resulted in the Arduino Core throwing a WiFi disconnect event to Homie. The WiFi reconnect does not work then at all and the Sonoff Basic keeps blinking in WiFi connecting mode.

I used a simpler solution to solve the issue for me: https://github.com/kleini/sonoff-basic/blob/master/src/main.cpp#L48. The event handler just tells the Arduino Core to properly disconnect. Homie afterwards tries to reconnect on its own.

Can some of the developers shed some light in this issue? Do you have any ideas, why first disconnecting and then reconnecting solves the WiFi connecting problems?

I am currently using
- platform espressif8266 version 1.8.0 containing Arduino Core 2.4.2
- Homie V2.1

JorinL, 2019-01-22:
@kleini helped me to develop a check which helps me deal with that problem.

Hopefully it is no longer a problem for me at all.

https://gist.github.com/JorinL/d97f3f2d95b93de946832c38409356ef

euphi, 2019-01-22:
I can confirm this for #develop-v3 branch, too. (no suprise, becuase event handling is unchanged).

euphi, 2019-01-22:
Problem is that if Wifi is disconnected before MQTT disconnection is detected, the flag to reset the "notify about MQTT-flag" is not reset. Therefore, after reconnection of MQTT will not be notified about.

JorinL, 2019-01-23:
To make it complete in its explanation - it has two ways which it can go:

1. MQTT Blinking - but Wifi connected and MQTT also
2. MQTT Blinking - Wifi and also MQTT disconnected.

JorinL, 2019-01-28:
Got some new informations:

If my devices are flashing "missing mqtt connection" they are controlable but don't answer on the topics.
They receive the command and act like they should but are not giving back anything.

JorinL, 2019-01-28:
Err did not got that - thanks for pointing me to it :-D Testing it now.

euphi, 2019-01-28:
I merged the fix to develop  branch.

closed 2019-01-28 by euphi

euphi, 2019-01-28:
If the _mqttConnectNotified has not been reset, but MQTT connection is re-established after connection fail, the homie-esp8266 does not set the ready flag (line  117 of Bootnormal.cpp in #develop-v3 branch), so `SendingPromise.send()` results in error: "✖ setNodeProperty(): impossible now"

JorinL, 2019-02-01:
I included the fix from [#545](https://github.com/homieiot/homie-esp8266/issues/545) to my Homie 2.0.0 works completely fine now - without any workaround fix from my side.

Thanks a lot @euphi

JorinL, 2019-02-20:
Mh I must say, that I still can bring devices (SonOff Touch, Wemos D1, SonOff Basic) into that fast blinking state (MQTT blinking) and not recovering. Even with that fix applied.

Possible it is missing on some other place too?

Which Branch of Homie is "suggested" to use? I'm currently on 2.0.0

commit 28fd2aed (2019-01-22): Try to fix #545.

It is untested, but I think if the MQTT disconnected event is missing,
e.g. due to loosing WIFI connection before detecting the MQTT
disconnect, the flag to notify about MQTT connection is not reset. Thus,
even if MQTT reconnects, the blinker is
commit c4c0fb05 (2019-01-28): Try to fix #545.

It is untested, but I think if the MQTT disconnected event is missing,
e.g. due to loosing WIFI connection before detecting the MQTT
disconnect, the flag to notify about MQTT connection is not reset. Thus,
even if MQTT reconnects, the blinker is